libretro/Lakka #588: 8BitDo Controllers won't reliably reconnect after being turned off and on

Lady-Luck, 2016-12-23:
Take your 8BitDo Controller, I have an NES30 Pro updated to the latest firmware, and do the connect and trust procedure that is described on the wiki.

Without powering the system off, turn off the controller. Now turn it back on again. Now turn it back off again. Now turn it back on again. Now turn it back off again. etc.

It will eventually fail to connect at all, even after a reboot.

This is with the December 22nd update of Lakka.

Alphascrypt, 2016-12-23:
Same Problem too. I cant connect my NES30PRO not via commandline. Nothing. Raspi3 Update 1222

ado464, 2016-12-26:
Hello! Did you run the "pair" command after following the wiki guide? That solved it for me, and it always connects. So "pair MAC-Adress" and that should be that. Hope it works for you as well

Best regards,
Ado

Lady-Luck, 2016-12-26:
Yes, it wouldn't let me, it told me it was already paired.

CoalaJoe, 2016-12-31:
I have pretty much the same problem.
I cant easily reconnect after reboot, but I can re-pair it via CLI.

lesavonfou, 2017-01-06:
Same problem here as well!
I can't test it on the moment but does the controller use the same bluetooth address?

fr500, 2017-01-06:
It's not lakka's problem it's 8bitdo's.
Same happens on windows.

Lady-Luck, 2017-01-06:
Does not happen with retropie.

fr500, 2017-01-06:
It happens to me in vanilla ubuntu with nothing RetroArch related running... sometimes the controllers fail to reconnect.

Lady-Luck, 2017-01-06:
Are your controllers using firmware 1.71?

woest, 2017-01-06:
Judging solely from their support forum, it _seems_ like they are working on the "automatic bluetooth connection" in their v1.72 beta firmware

http://forum.8bitdo.com/thread-1179-1-1.html#pid4032

Have not tried it yet.

Lady-Luck, 2017-01-06:
Doesn't really sound like the problem we're having.

fr500, 2017-01-06:
@Lady-Luck I updated last week. The latest firmware is the worst in my experience, they make it flicker rapidly when trying to connect (quite annoying on the NES30 Pro) and then it just seems to stop looking. I have to turn them off/on a few times to get them to connect.

Lady-Luck, 2017-01-06:
Well it's working on retropie and it's working on Windows so *shrug*

fr500, 2017-01-06:
I'm on windows and I'm having problems with reconnection with one pad (but not other 3 pads)

CoalaJoe, 2017-01-12:
Did anyone try the beta drivers?

Kivutar, 2017-01-19:
I could reproduce this with my RPi3 and my NES30Pro but not my SFC30.

Tested both on Mode 1.

farwrc, 2017-01-27:
I don't know if is this true but i tested on my RPi3 and zero and works, 
When you pair the controller successfully for the firs time and them turn it off when you turn back on, the leds blink like mode 4, but if you press A button for 5 seconds, the led blinking like mode 1 and reconnect to the console, i try this for several times and works, in lakka i am not tested yet but i thing is the same 

RPi3 with retropie
RPi3 with recalbox
RPizero with retropie
RPizero with recalbox
RPi3 With lakka will test later

My Firmaware 1,71 controller FC30 Pro 

regards
from brazil
hope to help all

Jaygatesny, 2017-02-09:
Just spent 3 hours trying to get my FC30 Pro to reconnect after shutdown on my Pi Zero. My problem is I cannot get the FC30 to connect from a shutdown. For instance, if I shut down the pi and controller, it will not connect when starting up, But if repair the controller, do a pi reboot and leave the controller ON during startup it will reconnect perfectly. So every time I want to use the controller from startup I have to unpair the controller, delete the device, rescan and add the device as new. I updated the FC30 to the most recent firmware on the 8bitdo website. I followed the guide completely on the wiki page and even added the script in udev/rules. One thing I noticed different in pairing mine multiple times is I'm never asked for "input pin 0000" and I think that could be my problem. Any help would be greatly appreciated.

Jaygatesny, 2017-02-10:
Got it working! My problem was my PC's Bluetooth dongle was Interfering with the FC30 Pro! I had to turn off "allow devices to connect to PC" and now it connects my Pi Zero  every time. I don't know why if the FC30 Pro is paired to my Pi that it wants to try and connect to my PC but that's what it was doing and took a lot of trial and error to figure this out. Hope this helps someone out there!

megacide, 2017-04-11:
Same problem here 8bitdo will not reconnect on reboot and power off of the controller.  When I check the config is saved after a reboot but the 8bitdo never reconnects.  I tried recent firmware- 2.0 was just released and still doesn't reconnect.  Been trying to solve over a week.  Giving up on Lakka  - it works on everything else but Lakka.

Jaygatesny, 2017-04-11:
My issue is fixed. Problem was bluetooth interference. These controllers are super sensitive to other devices broadcasting signals near by. My culprit was my PC's Bluetooth dongle I use for my TV remote. Every time I want to use my 8bitdo controller I have to turn off "allow bluetooth devices to connect" on my PC. This took days of trial and error and total mind fuckery to figure out. Hope this helps!