libretro/Lakka #589: Lakka on TVbox MXQ s805

germinbmxer, 2016-12-23:
Hello all! I'm Brazilian and don't write english very well, but I will try to explain the issue.
I'm trying to run lakka on TVbox MXQ S805 (the hardware is very similar to Odroid C1), but I don't have much knowledge.
Ok, I boot the system using Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf.img by putting manual commands to test like fatload, bootm, etc, and using DTB file of TVbox because DTB of Odroid on this image doesn't work. The system starts, work pretty well, connect on wifi, etc, but after 10 minutes +/- the display turn in Black image like a suspend mode, but the game is running normally(i hear the sounds in the game because background music and when I do commands on joystick like a jump, etc). This problem occur in any situation, playing the game or in lakka's menu. Not problem of temperature, because i put a dissipator and the CPU temp is around 38 degree C. No problem on the board because I tried with another one. I don't know why the video image disappear.

 I tried to use a new image file to create my SDcard, but After this version(Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf) the systems doesn't boot! The TV stops with Lakka Logo. 
The messages on serial stops in:

[  OK  ] Started Avahi Daemon.
[  OK  ] Stopped Udevil mount service.
[  OK  ] Stopped Udevil mount service.
[  OK  ] Created slice system-udevil\x2dmount.slice.
         Starting Udevil mount service...
         Starting Udevil mount service...
[  OK  ] Reached target Sound Card.
         Starting Samba NMB Daemon...
[  OK  ] Started RPC bind service.
[  OK  ] Started Samba NMB Daemon.
[  OK  ] Stopped Samba SMB Daemon.
         Starting Samba configfile writer...
         Starting WPA supplicant...
[  OK  ] Started Samba configfile writer.
         Starting Samba SMB Daemon...
[  OK  ] Started Samba SMB Daemon.
[  OK  ] Started Udevil mount service.
[  OK  ] Started Udevil mount service.
[  OK  ] Started WPA supplicant.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.
         Starting RetroArch user autostart script...
[  OK  ] Started RetroArch user autostart script.
         Starting Retroarch...
[  OK  ] Started Retroarch.
[  OK  ] Reached target Retroarch.

and nothing happens! The Lakka logo stays on TV.

I edited the bootargs to display console on laka using "Ctrl+Alt+F3" and do this command like described on Lakka page:
`systemctl stop retroarch.service; retroarch -v`

and system returns this, bub I don't know how to solve:
[image]

All images after "Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf.img" do the same screns!
The complete serial boot is present here:
[https://drive.google.com/open?id=1ov4SvGVdCOK3BNIcX2IseCg69URJPTbXPPG_EE6RIzQ](url)
Thank'you.

Kivutar, 2016-12-23:
Hi, the C1 image is not similar enough to run on your MXQ. Wait for the S805 port of lakka to be finished.

closed 2016-12-23 by Kivutar

germinbmxer, 2016-12-24:
Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf.img work on MXQ, but after 10 minutes the screen goes turn off

Kivutar, 2016-12-24:
Is the menu displaying before that?

germinbmxer, 2016-12-24:
lakka's menu?
With this image, the system work normally, i play the game normally, but after +/-10 minutes th screen goes off, but sound and commands are working (I know because the sounds, but the sreen doesn' return)

germinbmxer, 2016-12-24:
As you can see, Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf.img is running normally on MXQ without any mod. But after a time, screen turn black.
[image]
[image]

germinbmxer, 2016-12-24:
The time is exactly 10 minutes! I think that has a "suspend screen" or "screensaver" activated.

germinbmxer, 2016-12-25:
I made some new tests. If I maintained a USB keyboard plugged, when screens goes off, I press some key in keyboard and screens returns!

Kivutar, 2016-12-25:
you need to add a consoleblank=0 in the cmdline maybe

germinbmxer, 2016-12-26:
Worked! Thank you. I add `consoleblank=0` on cmdline like you described.
Now, I using Lakka-OdroidC1.arm-devel-20161002115148-r21593-g03241bf.img with another DTB archive in MXQ tv box running perfect. I tried to update to last one, but appear message md5 error :(
But for me, is ok to run this version.

Kivutar, 2016-12-26:
Ok good to know. Any idea if your box is one of these?

    HD18Q  m201d  MK808B.Plus  MXQ

germinbmxer, 2016-12-26:
I don't know what my box, but .dtb file of all of these versions boot here normaly(only with OdroidC1.arm-devel-20161002115148  or old versions before this):
`meson8b_hd180.dtb` `meson8b_mk808.dtb` `meson8b_mxq.dtb` `meson8b_m201.dtb`
I only need to rename files to `meson8b_odroidc.dtb` and replace in SDcard.

germinbmxer, 2016-12-26:
if you need to see these files, i put here:
[https://drive.google.com/open?id=0B-kAdjgbtkY6VzFmdnY4eE9jQXc](url)

germinbmxer, 2016-12-27:
I found this image: [static.kivutar.me/Lakka-S805.arm-devel-20161129115602-r21683-gb135373.img.gz](http://static.kivutar.me/Lakka-S805.arm-devel-20161129115602-r21683-gb135373.img.gz)

It working too

Kivutar, 2016-12-27:
Yeah it's the image we are currently working on. So you have an HD18Q like me.

germinbmxer, 2017-01-06:
Nice notice. :)
will you create instalation to NAND?

olekhov, 2017-01-18:
Greetings, I'd like to get involved.
I've got MXQ clone, black square box 11.5x11.5cm with red-blue LED and MXq label.
Rear connectors: 
USB-4
SPDIF
AV (with button accessible with a match or toothpick through the jack hole)
HDMI
Ethernet
Power connector 5v barrel

Boot logo says i'ts MXQ HD18Q mbox  v2
[image]

wifi chip is Realtek RTL8188ETV

motherboard says it's v1.1

The core board has an UART, how convenient; soldered a header there. UART is 115200 8N1
[image]

Jumped to UBoot shell, some logs are attached.

[uboot.txt](https://github.com/libretro/Lakka/files/713218/uboot.txt)
[mmcinfo.txt](https://github.com/libretro/Lakka/files/713219/mmcinfo.txt)
[env.txt](https://github.com/libretro/Lakka/files/713220/env.txt)

Tried to manually boot r21683 image from previous comments. The boot process starts normally, but then ends up complaining missing /flash/SYSTEM
[boot.txt](https://github.com/libretro/Lakka/files/713221/boot.txt)
[image]

It's a pity that serial console is disabled. Hooked USB keyboard and got some access to system.
[image]

How to properly boot Lakka?

I'd like to get dualboot with Android, if the SD with Lakka is missing. I think some twiddling with "preboot" or "recovery" uboot environment vars will do the trick.

Kivutar, 2017-01-18:
It's just an HD18Q, we support it, try this image http://static.kivutar.me/Lakka-S805.arm-devel-20161129115602-r21683-gb135373.img.gz