libretro/Lakka #593: Neo Geo screenshots not displayed at all

killer101, 2016-12-25:
Hi! All my screenshots for all other systems are displayed without any problem. The Neo Geo (Tab) want display screenshots at all. I named all of them to the playlist names. Nothing at all. Any tricks to do so, or just a bug!?

Kivutar, 2016-12-25:
Hi, we don't have a Neo Geo tab in lakka, but we have a tab for FBA

killer101, 2016-12-26:
Interesting! So why do I have one? I made a playlist for Neo Geo. The Controller Symbol is shown correct as tab. The Thumbnail Updater has a package for it. Everything is running smooth except the screenshots. Any reason for this?

Kivutar, 2016-12-26:
Try to enable GPU screenshots in the video settings?

markwkidd, 2016-12-26:
@killer101 there are currently no screenshots for Neo Geo in the thumbnail repository, just boxart.

You can probably copy the correct screenshots from the MAME thumbnail set into the Neo Geo folder on your Lakka installation (by creating a folder called Named_Snaps) in order for them to appear.

You could also try to use the AHK Arcade Playlist and Thumbnail script to download just the correct Neo Geo snaps and title screens if you have access to a Windows PC and some time on your hands: https://libretro.com/forums/showthread.php?t=7802

killer101, 2016-12-27:
I have my own screenshots and everything else set up for the Neo Geo tab. Lakka isn't just displaying them. They work perfectly if I put them to the FBA tab!

Kivutar, 2016-12-28:
Show me the path of your Neo Geo screenshots?

killer101, 2016-12-28:
I use "SNK - Neo Geo" ... "Named_Snaps". Like the one the Thumbnailupdater creates.

Kivutar, 2016-12-28:
And is your tab also named "SNK - Neo Geo"?

Kivutar, 2016-12-28:
In our RDB, I see no SNK - Neo Geo https://github.com/libretro/libretro-database/tree/master/rdb

killer101, 2016-12-28:
No, the tab is just named "Neo Geo". The rom symbols are CDs. The Thumbnail Updater is providing material for "SNK - Neo Geo". You are right about the rdb. I think something is messed up here. ...Maybe Lakka can provide a seperate tab for Neo Geo in the future. I like to have the roms seperated from the FBA stuff?!

killer101, 2016-12-28:
No, if you do so you end up in a tab without symbols. Just black squares!
But I found the solution ...
The name of the lpl has to be "Neo Geo.lpl", also inside the playlist. Example attached at the end!
The name of the thumbnail dir has to be "Neo Geo" too.
Now you get a controller symbol, rom symbols and the games are working. Maybe we have an unoffical Neo Geo tab now. Looks good, runs fine, works for me!

Maybe this tab was officially reserved for the "Neo Geo CD System" as it shows CDs as rom symbols?!

Here's the example ...

/storage/roms/neogeo/blazstar.zip
Blazing Star
/tmp/cores/fbalpha_libretro.so
FB Alpha
DETECT
Neo Geo.lpl

closed 2017-02-25 by killer101