libretro/Lakka #594: Problem with Wi-Fi on SHUTTLE Barebone XPC nano NC01U Celeron 3205u

shakkalakka, 2016-12-27:
Barebone PC has issues with W-Lan connection. all seems to be good at first.  Wi-Fi scanning runs and i can connect to the wi-fi of my choice but samba and internet doesnt run. If i make a rescan there is only one cryptic w-lan name (for example a reversed "?"-symbol)
W-Lan adaptor: Realtek RTL8821AE  

has try Lakka OE  (last 2 stable with 32Bit and 64Bit also in live-mode and legacy installation) and an older Lakka LE release (32Bit/64) 

all results in the same issue

Kivutar, 2017-02-06:
Certainly because we don't have RTL8821AE : https://github.com/libretro/Lakka/tree/lakka/packages/linux-drivers

shakkalakka, 2017-02-25:
any change the driver will packed in for future ?