libretro/Lakka #596: ScummVM Virtual Keyboard

killer101, 2016-12-28:
Hello! Is there any chance to bring up the virtual keyboard in ScummVM? Normally this is done by pressing the menu button 5 sec long. Doesn't work in Lakka. Or any other possiblity to enter letters by the controller?

DeFrenZ, 2017-01-18:
I have the same issue

killer101, 2017-01-22:
I just found this at the scummvm wiki ...

http://wiki.scummvm.org/index.php/Virtual_Keyboard

Seems to be that the virtual keyboard has to be enabled while compiling. Maybe a programmer/builder of Lakka can give it a try at the next release! Would be very nice to have this running!

killer101, 2017-02-18:
Any thoughts on this so far?

gouchi, 2017-02-19:
Please open [issue](https://github.com/libretro/scummvm/issues) on dedicated repo.

killer101, 2017-02-26:
I opened an issue there.

Maybe it is not a big problem except in Lakka. Retroarch for Windows passes keypresses from the keyboard to the game. Lakka doesn't. An option for this would do the trick for me too. The Virtual Keyboard would be the nicer solution by far though.