libretro/Lakka #597: FBA samples folder

killer101, 2016-12-28:
Can't get FBA to find the relating samples to the roms? Where do the samples folder have to go to?

shakkalakka, 2016-12-29:
i have the samples(*.zip) stored in "./system/fba"
(the "fba"-folder have to created at first)

dont know exactly but it seems to work

killer101, 2016-12-29:
Thanks, works perfect!

killer101, 2016-12-29:
Maybe someone can put this info to the docs. I think a lot more people are interested.

hericstrada, 2017-01-04:
Sorry, but, what is those samples?

shakkalakka, 2017-01-05:
the samples are mostly additional audio-files for a few acade-games. Some original PCBs uses "real audio" instead of "chiptunes" (as i understand it)

juri74, 2017-01-18:
hello thanks very much!!
i didn't know about that, so is the final path "storage/system/fba/samples/"? (and "storage/system/mame2003/samples/"? for mame2003?) all lowercase?
one question can i have both FBA and MAME2003 collections? does they appear in different list or one big list in lakka menu?
thanks again!

shakkalakka, 2017-01-20:
hello,
**"storage/system/fba/samples/**? (?=*.zip) 
and 
**"storage/system/mame2003/samples/**? (?=*.zip)
are the right paths.

Mame2003 and FBA Files are shown as seperate Playlists. The PL for FBA can be generated by lakka. 
for mame2003 you have to use a 3rd Party Playlist Generator (i recommend markwkidd's "AHK RetroArch Arcade Playlist and Thumbnail Generator").

Kivutar, 2017-02-05:
So can we close this?

killer101, 2017-02-18:
Yep, think so!

closed 2017-02-25 by killer101