libretro/Lakka #598: Screenshot to thumbnail (Just an idea)

hericstrada, 2016-12-30:
Sorry if this is not the place, but I just want to know if it's possible in the future, to have an option to take an Screenshot and it is automaticaly defined as the Screenshot thumbnail of that game in that playlist, all inside Lakka.

Thanks and have a great New Year!

cbluth, 2016-12-30:
A feature that I like to use is the "screenshot savestate" feature. It saves a screenshot everytime you save a savestate, it helps me remember which savestate I last used.
Also for the main boxart you can replace the thumbnail with whatever png-formatted file you want.

Kivutar, 2017-01-01:
This feature is there but causes problems with some GL cores.
You can still try it in the Advanced Settings, in Saving Settings.