libretro/Lakka #601: Live USB mode with efi images

apesterin, 2017-01-02:
I am trying to use efi images from the nightly folder - e.g. [Lakka-Generic.x86_64-devel-20161221183631-r21729-g274b40f-efi.img.gz](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161221183631-r21729-g274b40f-efi.img.gz)

However, unlike the non ufi build I get from downloads page, the efi one doesn't let me boot Lakka directly from the USB. It only gives me the option to install Lakka to another device

Is there an EFI Live USB image?

Kivutar, 2017-01-03:
Edit the bootloader config in the EFI folder. You can add the live options. Take example on the normal bootloader config which is on the same partition.

Kivutar, 2017-02-06:
Ok, try the libreelec based images, the live mode is now called "run mode" http://le.builds.lakka.tv/

Kivutar, 2017-03-05:
This bug won't fix on this repo. But we do support this live mode on EFI in the new LE based images.

closed 2017-03-05 by Kivutar