libretro/Lakka #602: Lakka hangs randomly

apesterin, 2017-01-02:
Running Lakka-Generic.x86_64-devel-20161221183017-r21729-g274b40f.img.gz
via Live USB mode on a Zotac CI323 nano mini-pc (Intel n3150)

I get random freezes when trying to start a rom, or scanning ROMs. The only way out is a hard reboot

Attached:
[system.txt](https://github.com/libretro/Lakka/files/680458/system.txt) - details of my system
[log.txt](https://github.com/libretro/Lakka/files/680457/log.txt) - retroarch logs, up to the hang (this one was when starting a n64 rom. the rom loaded upto the game start screen, and after I pressed start on the controller, the screen went blank and the system hung)

The behaviour isn't unique to this rom, and happens during other roms / activities (like scanning ROMs etc.)
Any help or pointers on how to move forward would be appreciated

shakkalakka, 2017-01-02:
had exactly same issues with a intel NUC (N3700 CPU)
seems to be a problem with the "Braswell" Architecture 

only change to Lakka-LE (Test) helps me

apesterin, 2017-01-08:
Any ideas as to what difference with LE might be the cause? I am close to abandoning my pursuit of lakka on PC due to this issue. Was LE stable for you to stay there?

gouchi, 2017-01-10:
Lakka LE has more up to date kernel and mesa package.

You only reproduce it with n64 core ? Did you try with GLupeN64 core ?

shakkalakka, 2017-01-10:
for me its very stable after change to LE. no more freezes of any kind! Big Thanks to gouchi at this point ;)

Kivutar, 2017-02-05:
OK, since we are going to switch to LE soon, I close this.
The builds are there http://le.builds.lakka.tv/

closed 2017-02-05 by Kivutar