libretro/Lakka #603: USB SNES-Controller Adapter  -  D-Pad not working

shakkalakka, 2017-01-02:
I would use my old Snes controllers with usb adapters i bought from ebay. 

I've tested them before on windows and they work perfectly.
However, when i try to map them on Lakka, i can map every button except D-Pad down and D-Pad Right. Any ideas on how to fix this?

here is a pic. of the adapter : http://www.directupload.net/file/d/4579/t4mm5oea_jpg.htm

Has tried with:
Lakka-Generic.x86_64-devel-20161002103858-r21593
Lakka-Generic.x86_64-devel-20161209103132-r21701

Arnochocobo, 2017-01-02:
The devs can't own any single piece of chinese adapters, and therefore, support might be "random". Just use a XBOX 360 controller, or get an everdrive if you really want to use your original controllers.

shakkalakka, 2017-01-02:
D-Pad on the X-Box 360 Pad isn't much better than a Original SNES-Controller connected to the cheap chineese usb-adapter (...even with the described issues^^)

gouchi, 2017-01-02:
Can you try append to [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc)
 
`usbhid.quirks=0x0925:0x8866:0x040`

If doesn't work try with 

`usbhid.quirks=0x0925:0x8866:0x008`

Thank you.

shakkalakka, 2017-01-03:
have tried booth cmdline parameters (exactly made like in the "cmdline-guide"). http://www.lakka.tv/doc/The-bootloader/#on-generic-pc

booth variants with same result...
D-Pad Down & D-Pad Right did not work in the Input Settings (all ohter axis-directions & buttons works fine).

gouchi, 2017-01-03:
Can you try to make a test with Settings Tab > Driver > Joypad Driver to linuxraw ?

Thank you.

shakkalakka, 2017-01-04:
have changed to linuxrax-driver (joypad). but D-Pad function is still broken.

shakkalakka, 2017-01-21:
any other suggestions?

hansen-m, 2017-03-21:
I recently bought a MayFlash SNES dual adapter and the D-Pad works at the Lakka menu but not in game. Is this a related issue? Although I'm not positive it is configured correctly since it appears to be incorrectly detected as the MayFlash N64 adapter.

https://amazon.com/SNES-Controller-Adapter-PC-USB/dp/B002IXZ5DE/

shakkalakka, 2017-04-02:
the adapter from amazon you post is another one.
i have booth adapters! 
the mayflash 2-port adapter i tried before buying the chinese 1-port adapter (sucks on D-Pad) ,the mayflash thing has exactly same issues on my system as you discribed (will detect as N64 Adapter and some buttons not working). But you can Bind all key correctly in input-settings, i think.

gouchi, 2017-04-02:
I forgot something when you try linuxraw driver with Settings Tab > Driver > Joypad Driver to linuxraw, you need to disable autoconfig via Setting > Input > Autoconfig Enable > Off and reboot. Bind it if necessary Setting > Input > Input User 1 Binds > Bind All.

Moreover, it will be great if you can can test with [Lakka LibreELEC RC4](http://www.lakka.tv/articles/2017/03/27/lakka-20-rc4-is-out/).

Thank you.

[Source](https://forums.libretro.com/t/mayflash-n64-adapter-support/5808/16)