libretro/Lakka #605: Lakka is

Anunknownentity, 2017-01-04:


Anunknownentity, 2017-01-04:
So I have an old Dell inspiron b130 laptop that I want to play games with. But after the flower screen the menu is so laggy I cant navigate through the settings to lower the resolution. So my question is, isn;t there a way to change the settings to the lowest before boot up so I can navigate through the menu? If not what can you suggest I do?

PS: b130 cant upgrade gpu, it can only upgrade ram and proccesor ( Will both of this help?)

Plz help me because I have no other uses for that laptop

gouchi, 2017-01-05:
Try to [disable ribbon](http://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow), if it is not enough try to change menu driver to rgui (menu_driver = "rgui").

You will have to edit the [configuration manually](http://www.lakka.tv/doc/Configuring-Lakka).

Arnochocobo, 2017-01-06:
Feel free to purchase something not being 23 years old, that might help too.

Kivutar, 2017-01-06:
Disabling the ribbon will certainly fix your problem.

It uses some functions not present in some video drivers.

closed 2017-02-06 by Kivutar