libretro/Lakka #606: [info] [Odroid XU4] can't access tty: job control turned off

cibersheep, 2017-01-04:
Just in case is usefull to someone else:
Yesterday I tried to flash several times a microSD for my Odroid and I got stuck in first boot with error «can't access tty: job control turned off». I thought it was the image and I download it again. Same result.

Just a tip: I had my eMMC on the Odroid at the same time I was tring to boot from microSD. Taking eMMC out solves the problem.

I'm happy. Good work with Lakka!

Kivutar, 2017-01-04:
We often get this error when the file system is corrupted. Reboot your laptop, redownload the image, reflash the SD card, and it will certainly work.

cibersheep, 2017-01-04:
Thank you, is a good advice, but I have tested booting the same image flashed on the same microSD:
I have observed that if the eMMC card is attached to the Odroid, at the same time that I try to boot from microSD, I get this error. Pulling eMMC card out and booting from microSD works.

closed 2017-02-05 by Kivutar