libretro/Lakka #610: Documentation: add BIOS information for Phillips Videopac+

markwkidd, 2017-01-08:
Earlier today we were able to merge in details for the Phillips Videopac+ BIOSes into libretro-dats, the O2EM core, and libretro-super.

There are three distinct Phillips Videopac+ BIOSes that should probably be added to the Lakka BIOS documentation. Here are there checksums, filenames, and proper names:
https://github.com/libretro/libretro-super/blob/e8310a1f507d6eb99b322916d60e35e52150ee2b/dist/info/o2em_libretro.info#L17-L25

This can take the place of the Odyssey2 BIOS information in that slot with a question mark that I added a few months ago as a placeholder. (I'm glad to finally know the answer to that question after so long :) )

Kivutar, 2017-01-15:
Hi @markwkidd 

I invited you as a collaborator on this repository so you can take care of this BIOS page yourself.

Don't abuse this right, I closed the wiki last year because people wee adding too much useless or false info.

Kivutar, 2017-02-06:
Can we close this?

markwkidd, 2017-02-10:
I'm sorry I lost track of this message @Kivutar . I meant to write you before now, because I do not seem to have access to the BIOS page to make the edit myself. At least there is no "Edit" button visible for the Lakka BIOS wiki page.

I would still be glad to make the update if you can help me access the editor.