libretro/Lakka #611: Programming languages

CoalaJoe, 2017-01-08:
Is it possible to install python, php etc. in lakka? I'd like to develop an API and all I see are shell scripts.

Kivutar, 2017-01-15:
For now i'm against this idea. I want Lakka to stay Linux+RetroArch+Cores.

However you can compile your own version of lakka and enable python which is already packaged.

What kind of tools would you like to develop?

CoalaJoe, 2017-01-16:
APIs.
Like a webInterface for playlist generation.
Etc.

Kivutar, 2017-01-16:
About playlist generation, the time would better be spent finalizing it in RetroArch. Don't you think?
Do you have other examples?

CoalaJoe, 2017-01-16:
API  for HomeAutomation
Upload of ROMs
WebInterface for Configuration editing
Backup and Restore

Kivutar, 2017-01-16:
Thanks

> API for HomeAutomation
Can you explain me what it will allow us to do?

> Upload of ROMs
There are already many ways to upload roms:
  * SAMBA
  * SCP
  * SSHFS
  * plugging an USB stick and loading roms from there
  * mounting NFS or CIFS
  * Downloading ROM through RetroArch directly by changing the assets_url
  * Using a Lutro app with LuaSocket

> WebInterface for Configuration editing
This one is interesting and has been requested another time. However, I don't get why it is easier to configure Lakka through a web interface rather than just using your joypad on the console itself. Also we would need to stop RA the time we save retroarch.cfg, which would stop the current game.

> Backup and Restore
You can already do this with:
  * SAMBA
  * SCP
  * SSHFS
  * plugging an USB stick and loading roms from there
  * mounting NFS or CIFS
  * And i'm going to package rsync

CoalaJoe, 2017-01-17:
> Can you explain me what it will allow us to do?

Examples: 
- When you start a Zelda ROM, your Google Cast speaker will play a short melody.
- When starting a ROM, your Philips Hue lights will change color depending on the genre or cover of the game.
And many more.

> There are already many ways to upload roms:
> 
> SAMBA
> SCP
> SSHFS
> plugging an USB stick and loading roms from there
> mounting NFS or CIFS
> Downloading ROM through RetroArch directly by changing the assets_url
> Using a Lutro app with LuaSocket

Sure, but a lot of people that just want to play games don't know what SAMBA, SCP, SSHFS, NFS is.

> This one is interesting and has been requested another time. However, I don't get why it is easier to configure Lakka through a web interface rather than just using your joypad on the console itself. Also we would need to stop RA the time we save retroarch.cfg, which would stop the current game.

It is easier to type on a physical keyboard rather than the joypad. Mostly you won't play a game AND do config, and even if. If it is clarified, it shouldn't be a problem.

> You can already do this with:
> 
> SAMBA
> SCP
> SSHFS
> plugging an USB stick and loading roms from there
> mounting NFS or CIFS
> And i'm going to package rsync

Just as I said. Many people that want to play games don't know the most of this possibilities.

Another usage would be a bundled controller client for mobile phones.

Kivutar, 2017-01-17:
> Examples:
> 
>     When you start a Zelda ROM, your Google Cast speaker will play a short melody.
>     When starting a ROM, your Philips Hue lights will change color depending on the genre or cover of the game.
>     And many more.

Ok I didn't know!

> Sure, but a lot of people that just want to play games don't know what SAMBA, SCP, SSHFS, NFS is.

That's what the http://www.lakka.tv/get tutorial is about: showing them that a samba folder magically appeared in their file explorer and that they can just drop their ROMs in there.
I don't see in what teaching them how to access a local URL in their browser and upload the ROM through a web interface would be more user friendly.

If we want something really user friendly for the games, we code a rom store that allows you to browse and download the games directly from archive.org.

> It is easier to type on a physical keyboard rather than the joypad. Mostly you won't play a game AND do config, and even if. If it is clarified, it shouldn't be a problem.

There are only 2 or 3 things retroarch ask you to type:

 * a wifi password
 * an optional netplay nickname
 * a netplay IP, and this is going to be replaced by something automated
 * your retroachievements credentials, in case you use this

Consoles like PS4 or XboxOne are doing this with an on screen keyboard, not a web interface.

> Just as I said. Many people that want to play games don't know the most of this possibilities.

When they read the installation tutorial, they see all the samba folders and can guess by themselves that copying the content of the "screenshots" folder somewhere is a backup of their screenshots.

> Another usage would be a bundled controller client for mobile phones.

RetroArch already supports this with a builtin core. I've never tested it yet, but it's known to work.

Anyway, what language would you use for this? Can you build a list of packages we would need to add in Lakka to achieve this kind of web interface? We will consider the inclusion of this depending on the number and the size of the dependancies.

I think the most lightweight thing would be something like this https://www.archlinux.org/packages/community/any/python2-flask/ which is already 6 dependancies.

Something like this would be good for me https://lwan.ws/

CoalaJoe, 2017-01-18:
> That's what the http://www.lakka.tv/get tutorial is about: showing them that a samba folder magically appeared in their file explorer and that they can just drop their ROMs in there.
> I don't see in [...]

Sure, wich is pretty nice and well done. But still, many people would feel more comfortable using a website.

> There are only 2 or 3 things retroarch ask you to type:
>
> a wifi password
> an optional netplay nickname
> a netplay IP, and this is going to be replaced by something automated
> your retroachievements [...]

Right. I remembered pressing up and down until I got the character I needed. But it seems like there is a onscreen keyboard now, right?

> RetroArch already supports this with a builtin core. I've never tested it yet, but it's known to work.
>
> Anyway, what language would you use for this? Can you build a list of packages we would need [...]

Personally, if I have a free choice, I use PHP 7 or PHP 7.1. Thats what I'm used to and build a lot of governmental applications for switzerland in a great team. 
For the webserver I thought about lighttpd which my private hoster uses.
For the connection I would use a bi-directional websocket connection for low latency commands.
For concrete packages, I need a little bit more time and check some of my servers.

But keep in mind, there does not have to be PHP or Python at the beginning. A "static" HTML5 would do it for the moment, beacuse there is already a API which I could connect from the browser.
https://github.com/libretro/RetroArch/wiki/Network-Commands

But there Is no bi-directional communication, so I could not implement rubmle/virbation.

fr500, 2017-01-18:
We have discussed whole archive.org thing a lot.
Thing is it's technically illegal. Yes they have a DMCA exception bla bla
bla. Eventually someone will take notice and then we'll see if the
exception holds. I wouldn't want my project to be tied with illegal content
distribution.

As for a web interface. I wouldn't use PHP. Actually I'd wipe PHP from
existence. :p
But I have no vote here, it's just my opinion.

CoalaJoe, 2017-01-18:
> We have discussed whole archive.org thing a lot.
> Thing is it's technically illegal. Yes they have a DMCA exception bla bla
> bla. Eventually someone will take notice and then we'll see if the
> exception holds. I wouldn't want my project to be tied with illegal content
> distribution.

I wouldn't want that either, even though it would be legal in my country. ;P

> As for a web interface. I wouldn't use PHP. Actually I'd wipe PHP from
> existence. :p
> But I have no vote here, it's just my opinion.

PHP isn't as bad as it used to be.

It is used alot to teach web development and the newcommers write their fcking spagetthi code.
But in professional maner, it is widely used, reusable, object-oriented, secure and fast.
And also C compliant. You could write functions in C and import them as a library in PHP to use them.

Kivutar, 2017-01-18:
It's true that PHP is improving, however compared to other dynamic languages that were already existing, it was always late. This article is very interesting about what people that come from these other languages dislike in PHP https://eev.ee/blog/2012/04/09/php-a-fractal-of-bad-design/

Anyway, PHP in Lakka is a no-no, like java. Even Perl that I love is a no-no.

In an embedded distro, we choose the smallest tools possible, with the smallest amount of dependancies and that is usually written in C and compiles fast and supports ARM. The solution you propose with PHP and an HTTP server is already too much.

Take a look at something like https://lwan.ws/, I have the hope that it has no dependancy. It is its own http server, allows dynamic scripting in Lua (and Lua is damn small and fast).

@fr500 I never said we have to include it in Lakka. It can be done by a third party team that we don't know in the form of a bash script / a Lutro app / a libretro binary and distributed not by us. But the users who want to pirate would know its existance fast enough. In fact, I would be surprised in something like this doesn't happen when Lakka becomes famous, because it happenned to OpenELEC.

Like Firefox is not responsible of people streaming movies illegally, RetroArch is not responsible if people are configuring it to download games instead of icons and stuff.

Ah, and yes, we have an OSK now.

CoalaJoe, 2017-01-18:
Well if this is the case, I can't contribute to the backend very much.

However, I could do the front-end, if there is a need to.

Python is a no-no aswell?

Alcaro, 2017-01-18:
Let me check how big those things are.

perl (Debian Jessie, amd64) is [17584](https://packages.debian.org/jessie/perl/perl) + [5097](https://packages.debian.org/jessie/perl-base) + [15108](https://packages.debian.org/jessie/perl-modules) = 37789 KB installed (plus another ~1.8MB of language-neutral dependencies that may be optional and/or present already; I didn't count glibc). That's a lot for an embedded system.

[Python](https://packages.debian.org/jessie/python/python) (same platform) is full of virtual/routing packages, such as 'python' depending on 'python2.7'; discounting them (and lang-neutral deps) leaves 3825+8375+360 = 12560 KB. A lot better than Perl, but it has about 11MB of dependencies, so the real size (assuming half can be trimmed away) is somewhere around 18MB.

[PHP](https://packages.debian.org/jessie/libapache2-mod-php5) has a dependency list longer than Python, including [two](https://packages.debian.org/jessie/libonig2) [different](https://packages.debian.org/jessie/libpcre3) regex libraries, random tangential stuff like sed, lsof and Kerberos, the Apache module depending on the CLI edition, Perl (via lsof), and some stuff I'm not sure what it's good for. Not counting the language-neutral dependencies or the CLI version, the sum is 10933KB; with the deps (except Perl), the sum is 37411KB, so let's halve that difference too and say about 24MB here.

[Lua](https://packages.debian.org/jessie/lua5.2) only needs 363KB, and its only dependencies are glibc (we have that already) and readline (Lua docs mention only needing an ANSI C compiler, so that one is clearly optional).

 

But RetroArch itself is [40MB](https://packages.debian.org/stretch/retroarch-assets), Lakka (RPi-20161221) is 574MB (as .img, which should be roughly equivalent to Debian's installed size), so another couple of megabytes will barely be noticed.

Therefore, I'd rather see Python than Lua. It and PHP are a lot more popular than Perl and Lua, and while PHP is improving, it still carries a lot of baggage from older versions; Python is a lot more loved. Adding Python would increase the image size by about 3%, probably 1-2% if we tune the config options. I consider that worthwhile.

(on a totally different note, someone slap the Debian packagers, they added every single input/audio/video driver as a hard dependency and labeled rarch as an SDL app)

Kivutar, 2017-01-18:
Thanks for the figures. (the RPi image is 574mb but half of this is empty space, the rootfs is 196M)

But if we're doing to add web into lakka, which I still think is overkill, we also need to compare the scripting languages including their web frameworks and the dependancies of these web frameworks.

The problem with attracting python devs in a project like this is that tomorrow we'll get 100 requests like "I want this python package", "I want this python binding", "I don't like to code without this template engine", "where is mariadb?" etc.

It's like C++ devs in the kernel.

So my stance stays the same : it's a game console, it already allows to play games. Adding web technologies in Lakka will bring only side functionnalities at the cost of a lot of bloat. So I'm already being very open minded in giving you a hope to include something like https://lwan.ws/

CoalaJoe, 2017-01-18:
How about an optional bundle. Like you downliad extra cores.

Kivutar, 2017-01-19:
Like OpenELEC addons? That's a way we never explored yet, but it's worth giving it a try, because it can have so many uses.
Basically, addons are compiled at the same time as the OS image, with the same compiler and optimization. Then I think we host them somewhere, and we create a menu in RetroArch to download them.
I don't know how they look on the system once installed. I never used it before.
I guess the best way is to try on LibreELEC, we install a plain LibreELEC, then we download an addon, and check how it affects the storage partition, to get how the whole thing works.

CoalaJoe, 2017-01-19:
Nice! Are you going to check this, once you have time?

Kivutar, 2017-01-19:
I'm currently busy with porting Lakka to LibreELEC here https://github.com/lakkatv/Lakka-LibreELEC
I'd like to close this current repository and switch to the new one before the end of february.

We can start looking at:
https://wiki.libreelec.tv/index.php?title=Add-ons
https://github.com/lakkatv/Lakka-LibreELEC/tree/master/packages/addons
http://addons.libreelec.tv/8.0/RPi/arm/

CoalaJoe, 2017-01-19:
I found a LAMP addon: https://forum.libreelec.tv/thread-280.html

Kivutar, 2017-01-19:
Ah well found. Now we need to understand how to install it manually, without KODI

SaBomNim, 2017-04-06:
Without python is there a way to add a power/reset button from the gpio to safely shut down or reset the system?

Kivutar, 2017-04-06:
Yeah using bash and the gpio binary

bparker06, 2017-04-07:
Any programming language/shell should be able to directly work with gpio by writing to /sys/class/gpio as well.

Kivutar, 2017-04-07:
Yes, if I understand well, wiringPi just sits on top of this to make it more "Arduino" like.

SaBomNim, 2017-04-07:
Can anyone post a link to a bash script that would work, I can't seem to get it to work for me.

Kivutar, 2017-04-07:
The doc is there, it comes with examples http://wiringpi.com/the-gpio-utility/