libretro/Lakka #616: ssh service does not match interface

EdwinKM, 2017-01-15:
After a fresh install the SSH daemon is running. The interface defaults to "off". But i can still login.

Kivutar, 2017-01-15:
Yes we force SSH ON in the cmdline.

The UI should somehow reflect that this switch has no effect when ssh is forced in the cmdline, by greying the switch.

EdwinKM, 2017-01-19:
i did not notice a grey state. A undefined state looks rather strange to me. Seems more logical to always show the right state in the menu?

Kivutar, 2017-01-19:
We need to show the switch on the right, but also show the user that he can't switch it OFF, because it's forced ON by the cmdline.

EdwinKM, 2017-01-21:
thanks for your fast replies, but i really do not see the problem. I suspect there are use cases i do not have?

From my perspective:
* burn sdcard and boot
* use monitor and see in the menu that the ssh server toggle is disabled
* use laptop and i can successfully ssh to my raspberry pi
( i did not test if a second boot it fix itself)

Most simple solution would be to change the default toggle to "on" in the menu. Or is this a separate project?  But options in a menu should always match with the real applied options.

As it is marked as "enhancement" i suspect you understand the issue. And it is not a real deal breaker. Thanks in advance

Kivutar, 2017-01-22:
The use case you forget is this one:

burn the SD card and boot
use monitor and see in the menu that ssh server toggle is enabled (because we fixed the bug)
toggle ssh OFF
ssh still works, because you never removed the ssh argument in the cmdline

So you end up with ssh working while the switch is off, even if we fix the bug.

We need to make that button impossible to turn off, if ssh in forced on in the cmdline.

And we need to give the user a visual clue that this switch is locked ON.