libretro/Lakka #617: Fatal system fail

luixal, 2017-01-15:
Strange this here. While running Lakka on a Rpi3 (fresh install, just wifi had been setup), I made the following steps in the config menu:

1. Enable show advanced settings
2. Disable show advanced settings

then the system crashed (it hanged) so I plug power off and back on again.

When booting it took some time and the showed a message about not being able to access tty and that it couldn't run the job.

No `retroarch` command worked. Had to reinstall sdcard again.

Kivutar, 2017-01-15:
Something unexepected made the system crash and the filesystem got corrupted. It happens in very rare cases. So rare that it's crearly impossible to debug it.
Thanks for reporting, but I doubt there is something I can do at my level about it.
With time, RetroArch uses coverity to hunt memory issues.
And with our next switch to LibreELEC, the kernel of the RPi will be updated too.
This kind of things are more likely to fix this kind of bug than me trying dive into it.
So I close this report.

closed 2017-01-15 by Kivutar

luixal, 2017-01-20:
Not so rare: reproduced it three times in a row and finally had to drop Lakka for now as a stable solution.

Any change I made to the menu (switch themes, show and hide advanced settings, etc...) ended up in black squares instead of icons (not every icon, randomly), theme going to Pixel although a selected another one, hanging after a really short navigation through the menu...

I think that switch to LibreELEC will be for good. I'll give it another try then :)

Kivutar, 2017-01-20:
Well, I'm in the menu right now, switching Advanced Settings ON and OFF like 30 times, and nothing happens. For now I've never been able to reproduce this bug.

> Any change I made to the menu (switch themes, show and hide advanced settings, etc...) ended up in black squares instead of icons (not every icon, randomly), theme going to Pixel although a selected another one, hanging after a really short navigation through the menu...

The only officially supported theme is Monochrome. All the other themes are maintained by occasional contributors.
Changing a theme requires restarting RetroArch.

The switch to LibreELEC is progressing well, I will post images soon 😃

luixal, 2017-01-20:
It's quite rare. I discard hardware problems as the Rpi3 si working fine with other distros and I flashed Lakka three times.

If I'm able to reproduce the error when I try it next time, I'll take a video to give you more information.

Cheers!