libretro/Lakka #621: Problem with dualshock3 on Odroid C2

ziogit, 2017-01-17:
Odroid C2 (Lakka december nightly) with wifi usb key, original ps3 controller, logitech K400 keyboard, BCM2035 Bluetooth dongle.
booting ok and keyboard running.
Using keyboard -> services -> enabled SSH, Samba, BT.
Pressed PS button, blinking, but controller is not working. Retry after a reset, nothing changed.
Tried a reeboot with no changes.
Dmesg output:

Lakka:~ # dmesg
[ 795.819117@1] usb 1-1.3: new full-speed USB device number 9 using dwc_otg
[ 795.829985@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 795.829985@0]
[ 795.841255@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 795.841255@0]
[ 795.852365@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 795.852365@0]
[ 795.929046@1] usb 1-1.3: device descriptor read/64, error -71
[ 796.182147@1] usb 1-1.3: New USB device found, idVendor=054c, idProduct=0268
[ 796.182162@1] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 796.182168@1] usb 1-1.3: Product: PLAYSTATION(R)3 Controller
[ 796.182173@1] usb 1-1.3: Manufacturer: Sony
[ 796.195289@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.195289@0]
[ 796.203563@1] sony 0003:054C:0268.0001: Fixing up Sony Sixaxis report descriptor
[ 796.215663@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.215663@0]
[ 796.226722@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.226722@0]
[ 796.237960@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.237960@0]
[ 796.249257@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.249257@0]
[ 796.249486@1] input: Sony PLAYSTATION(R)3 Controller as /devices/dwc2_b/usb1/1-1/1-1.3/1-1.3:1.0/0003:054C:0268.0001/input/input5
[ 796.249849@1] sony 0003:054C:0268.0001: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-dwc2_b-1.3/input0
[ 796.260882@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[ 796.260882@0]
[ 796.260929@1] sony 0003:054C:0268.0001: can't set operational mode
[ 796.369071@1] sony: probe of 0003:054C:0268.0001 failed with error -71

lsusb output:
Lakka:~ # lsusb
Bus 001 Device 007: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 008: ID 0a5c:200a Broadcom Corp. BCM2035 Bluetooth dongle
Bus 001 Device 010: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Kivutar, 2017-01-19:
I should be able to try this soon

ziogit, 2017-01-19:
Thank you very much.
Anyway if you need some tests don't hesitate to ask :)

jbaiter, 2017-02-22:
I have the same problem with my BCM2035 dongle, with the exception that it doesn't even show on `lsusb`:

```
Feb 22 20:45:55 LibreELEC kernel: usb 1-1.3: new full-speed USB device number 9 using dwc_otg
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: usb 1-1.3: device descriptor read/64, error -71
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: usb 1-1.3: device descriptor read/64, error -71
Feb 22 20:45:55 LibreELEC kernel: usb 1-1.3: new full-speed USB device number 10 using dwc_otg
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: usb 1-1.3: device descriptor read/64, error -71
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:55 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: device descriptor read/64, error -71
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: new full-speed USB device number 11 using dwc_otg
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: device descriptor read/8, error -71
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: device descriptor read/8, error -71
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: new full-speed USB device number 12 using dwc_otg
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: device descriptor read/8, error -71
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
Feb 22 20:45:56 LibreELEC kernel: usb 1-1.3: device descriptor read/all, error -71
Feb 22 20:45:56 LibreELEC kernel: hub 1-1:1.0: unable to enumerate USB device on port 3

LibreELEC:~ # lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Kivutar, 2017-03-05:
I was able to reproduce. But not yet to fix it. I close this bug report in favor of this new one: https://github.com/lakkatv/Lakka-LibreELEC/issues/54

closed 2017-03-05 by Kivutar

ziogit, 2017-03-05:
> favor of this new one: lakkatv/Lakka-LibreELEC#54

If this can help you:

Installed retropie from source (on Odroid Ubuntu official release)

There is a script (retropie_setup.sh) that allows to manually install ps3
drivers.

This procedure makes ps3 controller WIRED working.

Bluetooth connection not yet tested.

Hope this help

Sent by Mate 9