libretro/Lakka #622: Lakka on MXQ S805 v1.1

olekhov, 2017-01-18:
There is a closed issue #589, but my board is a bit older.

I've got MXQ clone, black square box 11.5x11.5cm with red-blue LED and MXq label.
Rear connectors: 
USB-4
SPDIF
AV (with button accessible with a match or toothpick through the jack hole)
HDMI
Ethernet
Power connector 5v barrel

Boot logo says i'ts MXQ HD18Q mbox  v2
[image]

wifi chip is Realtek RTL8188ETV

motherboard says it's v1.1

The core board has an UART, how convenient; soldered a header there. UART is 115200 8N1
[image]

Jumped to UBoot shell, some logs are attached.

[uboot.txt](https://github.com/libretro/Lakka/files/713218/uboot.txt)
[mmcinfo.txt](https://github.com/libretro/Lakka/files/713219/mmcinfo.txt)
[env.txt](https://github.com/libretro/Lakka/files/713220/env.txt)

Tried to manually boot r21683 image from previous comments. The boot process starts normally, but then ends up complaining missing /flash/SYSTEM
[boot.txt](https://github.com/libretro/Lakka/files/713221/boot.txt)
[image]

It's a pity that serial console is disabled. Hooked USB keyboard and got some access to system.
[image]

How to properly boot Lakka?

I'd like to get dualboot with Android, if the SD with Lakka is missing. I think some twiddling with "preboot" or "recovery" uboot environment vars will do the trick.

Kivutar, 2017-01-18:
Try this http://static.kivutar.me/Lakka-S805.arm-devel-20161129115602-r21683-gb135373.img.gz

olekhov, 2017-01-18:
The logs are from this very image.

I've managed to boot vanilla LibreELEC image from UBoot:

m8b_m201_1G#mmcinfo
[mmc_init] SDIO Port B:0, if_type=7, initialized OK!
Device: SDIO Port B
Manufacturer ID: 0
OEM: 0
Name: Tran Speed: 20000000
Rd Block Len: 512
SD version 2.0
High Capacity: Yes
Capacity: 15931539456
Boot Part Size: 0
Bus Width: 4-bit
m8b_m201_1G#

m8b_m201_1G#fatload mmc 0:1 ${loadaddr} aml_autoscript
reading aml_autoscript

702 bytes read
m8b_m201_1G#
m8b_m201_1G#autoscr ${loadaddr}

Your image lacks aml_autoscript file

loading just kernel and booting it from memory results in missing /flash/SYSTEM

olekhov, 2017-01-18:
Looks like LibreELEC tweaked uboot vars somehow.
This time image loaded just fine

olekhov, 2017-01-21:
The next problem is sound.
Nothing goes through HDMI (LibreELEC image works fine)

Lakka:~ # aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: AMLM8AUDIO [AML-M8AUDIO], device 0: AML PCM dummy_codec-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: AMLM8AUDIO [AML-M8AUDIO], device 1: SPDIF PCM dit-hifi-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

olekhov, 2017-01-21:
I've checked 3.5mm jack - the sound goes there.
How to route it to hdmi?

Kivutar, 2017-01-22:
Try to change the audio device using left and right in the interface.

Your TV is certainly sending a wrong EDID saying that it doesn't support hdmi audio.

But don't worry, you'll get LibreELEC based builds very soon, @Ntemis is activelly working on porting S805 to Lakka-LibreELEC.

olekhov, 2017-01-22:
There are only three options:
null
default:CARD=AMLM8AUDIO 
sysdefault:CARD=AMLM8AUDIO

`Lakka:/tmp # cat /sys/devices/virtual/amhdmitx/amhdmitx0/edid 
Receiver Brand Name: 
Receiver Product Name: 
EDID block number: 0x0
blk0 chksum: 0x00
Source Physical Address[a.b.c.d]: 0.0.0.0
native Mode 0, VIC (native 0):
ColorDeepSupport 0, MaxTMDSClock 0

Audio {format, channel, freq, cce}
Speaker Allocation: 0
Vendor: 0
Rx 3D Format Support List:
{VIC FramePacking TopBottom SidebySide}
`

my TV is actually a 24" monitor, BenQ E2400HD

olekhov, 2017-01-22:
The same edid file is for 55" Samsung TV (UE55H7000AT)
It seems that board can't get EDID from either display.

Kivutar, 2017-02-05:
We now have these images http://le.builds.lakka.tv/

Kivutar, 2017-02-17:
I fixed the audio issue yesterday. The builds are available here http://le.builds.lakka.tv/

closed 2017-03-05 by Kivutar