libretro/Lakka #623: Framerate drop in XMB on rPi3 with shader loaded.

originalpositron, 2017-01-19:
If I turn on the 'crt-pi' shader for any game, the FPS in the XMB drops from 60 to ~32-41.
This even persists when no game is suspended in the background. It also goes away as soon as I turn off the shader in a game configuration menu (set 'shader passes' to 0). 
The shader doesn't do anything to the XMB, so this should not be happening..

twinaphex, 2017-01-19:
No, this is normal. It is active still on the background game image when you enter back into the XMB.

originalpositron, 2017-01-19:
What background game image? This even persists when I explicitly close a game or reboot..

twinaphex, 2017-01-19:
Not a bug, you have this shader still running, hence if your GPU isn't up to the task, you get the 'frame drop'. Don't use a shader which your Pi 3 can't render at fullspeed, problem solved.

originalpositron, 2017-01-19:
Wouldn't it improve the UX to dynamically unload the shader when in the XMB, and load it again when going back into a game? Shaders load pretty much instantly so it wouldn't even be obvious to the user (aside from the much better framerate in the XMB)

originalpositron, 2017-01-19:
..why are you so hostile? ['crt-pi' is actually a shader specifically made for the (original!) Pi to run at full speed.](https://github.com/RetroPie/RetroPie-Setup/wiki/Shaders-and-Smoothing) I'm well aware that Lakka runs on multiple platforms.. all I'm saying is that it makes no sense to have the GPU doing any in-game shader work whilst the user is in the XMB. On decently powered devices this doesn't matter, but any low-powered devices will see XMB framedrops for virtually no gain.

twinaphex, 2017-01-19:
Maybe, maybe not. Anyway, it's not something I am going to spend time on in time for 1.4.0. For now, you should stop using shaders which your GPU obviously cannot handle.

originalpositron, 2017-01-19:
I didn't use the word easy anywhere, any time Something obviously put you in a bad mood today. Sorry, but I didn't cause whatever did that.
I'll try and hope you're calm and reasonable enough to grok what I'm trying to convene:

1. The XMB works at a fluid 60FPS normally
2. The 'crt-pi' shader runs at fullspeed on any Pi in any game, and was made for the (lot more) underpowered Pi1
3. Having any shader (even this carefully optimized one) loaded causes an immediate 25FPS drop in the XMB
4. This framedrop persists even past a reboot. The shader is probably loaded even without a game
6. This is not needed. Even when a game is suspended, the shader doesn't need to be active when in the XMB
7. To me this is the bug 'game shader is active even when in the XMB' or 'game shader active when no game is in the foreground'.

I can't see why you call this behaviour 'normal', as if its intended by the devs. There's no gain in it, and only a processing cost/FPS loss attached to it.

I'm not asking to have it fixed right away, its not usability breaking, but I thought it would be nice to at least have the devs take notice of it and have it filed  in the bug/issue tracker.

twinaphex, 2017-01-19:
I will take that into consideration as something to be looked at post-1.4.0.

originalpositron, 2017-01-19:
Cool, thanks!

Kivutar, 2017-01-20:
I never unsterstood why the menu was so slow on videocore. We might be doing something wrong with the shaders we use to draw quads, because writting text is very expensive on the Pi. For example, enabling icons and label shadows divides the FPS on the Pi, but not on the other ARM boards.

It's true that crt-pi is very optimized for the pi and should run without problem.

In fact, in Lakka, the default setting is to have libretro_pause ON, and a menu background opacity of 1. In this case, the game is never really displayed behind the menu, and we could add a test to avoid doing the shader passes, thus always have a fast menu. If it causes no other bugs.