libretro/Lakka #624: [FIX][X86_64] Making lakka x86_64 able to boot in (almost) any EFI configuration

menelkir, 2017-01-19:
It's seems changing the boot parameters to the example below, makes lakka x86_64 booting almost  every efi configuration. I think the problem is syslinux unable to auto-detect what framebuffer to boot (I think its a syslinux bug). Probably _video=_ should be revised depending on videoboard for a better performance.
I've tested _video=efifb_ with one machine with EFI+intel videocard, one machine with EFI+Nvidia videocard and a Macbook Air 2014. All working fine. 

```
LABEL installer
  MENU LABEL Run  Installer
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 installer tty vga=current vt.global_cursor_default=0 ssh video=efifb quiet 

LABEL live
  MENU LABEL Run  Live
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 disk=UUID=a738b422-ad59-4061-ae95-5f9636c646ce ssh vga=current vt.global_cursor_default=0 video=efifb quiet

```

For a better debug (in case something goes wrong and to be able to make a good report), you can add an extra option:

```
LABEL livedebug
  MENU LABEL Run  Live (Debug)
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 disk=UUID=a738b422-ad59-4061-ae95-5f9636c646ce ssh vga=current vt.global_cursor_default=0 video=efifb verbose ignore_loglevel earlyprintk=efi,keep
```

gouchi, 2017-01-19:
Thank you for the update !

It seems this [boot parameter](https://www.kernel.org/doc/Documentation/fb/efifb.txt) is only for EFI booted Intel Macs.

menelkir, 2017-01-19:
Tested with efi instead of efifb, also works with Intel Macs and regular EFI machines

```
LABEL installer
  MENU LABEL Run  Installer
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 installer tty vga=current vt.global_cursor_default=0 ssh video=efi quiet 

LABEL live
  MENU LABEL Run  Live
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 disk=UUID=a738b422-ad59-4061-ae95-5f9636c646ce ssh vga=current vt.global_cursor_default=0 video=efi quiet

LABEL livedebug
  MENU LABEL Run  Live (Debug)
  KERNEL /KERNEL
  APPEND boot=UUID=7f074201-a1d7-4091-8549-a78515a9d150 disk=UUID=a738b422-ad59-4061-ae95-5f9636c646ce ssh vga=current vt.global_cursor_default=0 video=efi verbose ignore_loglevel earlyprintk=efi,keep
```

Kivutar, 2017-01-20:
Looks like you're using a Lakka-OE image and not Lakka-LE. You're supposed to have a

    LABEL run

in Lakka-LE