libretro/Lakka #627: Problem with asrock j3455-itx

hackbird001, 2017-01-22:
Hello,

i installed both version 32/64 from the main on USB.
With both version boot hangs on the flower. Nothing happens.

My Board: Asrock J3455-ITX
Sound Card: Realtek ALC892
Graphics Card: Integrated Intel® HD Graphics 500: 12 EUs inside (Up to 750MHz)

gouchi, 2017-01-22:
Some [questions](https://github.com/libretro/Lakka/blob/lakka/ISSUE_TEMPLATE.md) haven't been answered ;-)

Which version of Lakka are you using ? Moreover, is it possible to get some [lakka system logs](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

Thank you.

hackbird001, 2017-01-23:
Hey, i used both version from the offical main page (lakka.tv)...
Log cannot post because it hangs on flover bootscreen

gouchi, 2017-01-23:
Please try to connect by [ssh](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#ssh) or [direct access](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#direct-access).

Thank you.

hackbird001, 2017-01-23:
direct access wont work.
How are the login?
I know u must use putty and my ip adress from htpc but password and so on?

gouchi, 2017-01-23:
If you can't ssh, you should have succeeded with direct access. 

Can you try to append to the cmdline 
`nosplash textmode`

It should disable the splash screen while booting and boot to a terminal.

hackbird001, 2017-01-23:
Hey gouchi,

never done before. 
Can u explain me how i should do this??

gouchi, 2017-01-23:
Please read [this documentation](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc).

hackbird001, 2017-01-24:
Hey gouchi,

wont work. did hangs on the flower bootscreen.

First i was type tab then it comes the cmd line with "kernel version text and so on" after this i type: nosplash textmode.....   Hangs on flower

Second i delete all the text and write again nosplash textmode.   Hangs again

gouchi, 2017-01-24:
Strange. Can you try to make a test with  [Lakke LE](http://le.builds.lakka.tv/Generic.x86_64/) (LibreELEC based) and select Run mode, if you reproduce the issue ?

Thank you

gouchi, 2017-01-24:
Strange it should have disable the splash screen. Can you try to append to the cmdline 
`debugging`

Thank you.

closed 2017-01-25 by hackbird001