libretro/Lakka #629: [suggestion] create symlinks in /storage/assets for content in /usr/share/retroarch-assets

markwkidd, 2017-01-22:
The goal is to make it possible to add custom icons and themes without requiring Lakka users to use SSH.

If there were symlinks for all of the content in /usr/share/retroarch-assets to make those files available in /storage/assets, then all a user would need to do is:

1. transfer their new theme content via samba
2. change the asset path in RetroArch to point to /storage/assets

It would not be quite as good as the proposed 'override' system described in the docs, but this seems like it would work pretty well in the meantime.

If you are interested in this @Kivutar but would like someone else to implement it, if you could point me to a starting place in the Lakka setup scripts I would be glad to try to make a PR. I couldn't decide where to try to make the change on my own!

Kivutar, 2017-01-22:
I'm not really sure it would work.

Even if it would, it's less powerfull than the current override system, that is needed anyway for joypad configs.

So I prefer to keep only one way to do overrides.

markwkidd, 2017-01-22:
I must confess I'm not familiar with how the current overrides work in terms of code.

I understand not wanting to make things confusing with another approach. 

Is there any chance you could point me towards the code that handles joypad overrides so I can see if I'm capable of extending that approach to artwork/themes?

Kivutar, 2017-01-22:
Sure, it's called overlayfs, take a look in the retroarch package. https://github.com/libretro/Lakka/tree/lakka/packages/libretro/retroarch/system.d

It already allows you to override assets.

markwkidd, 2017-01-25:
Thanks! I'm familiarizing myself with the folders with overlayfs.

Pardon a beginner's question, but it looks like there is already a configuration file to set up `/storage/assets` as an overlay to `/usr/share/retroarch-assets`
https://github.com/libretro/Lakka/blob/lakka/packages/libretro/retroarch/system.d/tmp-assets.mount

I don't see any difference between the way that file is setup and the configuration for the joypad overlay: https://github.com/libretro/Lakka/blob/lakka/packages/libretro/retroarch/system.d/tmp-joypads.mount#L11

Could you help me better understand where the missing piece is?

Kivutar, 2017-02-05:
There are no differences. Assets and joypads work exactly the same:

 * one folder in the squashfs is read only
 * one folder in /storage is rw
 * the resulting layered folder is in /tmp

markwkidd, 2017-02-10:
Ah, for some reason I had the impression that the overlay fs was not configured yet for assets. I will close this issue since it is moot. The benefit is that I have learned how the overlays are configured. Thank you!

closed 2017-02-10 by markwkidd