libretro/Lakka #630: No bootloader screen on LattePanda

tschirkyt, 2017-01-23:
Hi

I'm trying to run Lakka on my Lattepanda in USB live mode. But after the flower screen the system skips the bootloader screen and starts the installer, without offering the choice to "Run Live".

I'm using the latest nightly for x64 and UEFI (Lakka-Generic.x86_64-devel-20161221183631-r21729-g274b40f-efi.img.gz)

gouchi, 2017-01-24:
Yes for now with Lakka OE (OpenELEC based) we don't have the live mode for UEFI boot.

Is it possible to make a test with [Lakke LE](http://le.builds.lakka.tv/Generic.x86_64/)(LibreELEC based) and select Run mode ?

Thank you

tschirkyt, 2017-01-26:
Thanks for your reply. So it's just not implemented yet.
I tried to the linked Lakka LE (the build from 2017-01-21) and I get the error:

```
Failed to load COM32 file vesamenu.c32
boot:
```

then it repeats those two lines over and over. 

I just saw that there is a new Lakka LE build from yesterday. I'm gonna check this one out soonish.

tschirkyt, 2017-02-07:
Now I tried the latest Lakka LE build (2017-01-25), but with the same problem:

```
Failed to load COM32 file vesamenu.c32
boot:
```

gouchi, 2017-02-07:
If you press Enter or type run in the boot prompt nothing happened ?

Kivutar, 2017-02-17:
This error was due to us trying to load vesamenu.c32 in EFI mode. The new LE images have this bug fixed.

However, if I remember correctly, the guy on the forum who was able to boot on LattePanda was using BIOS emulation instead of EFI anyway. (Needs to be confirmed)