libretro/Lakka #635: Lakka don't sees the Xbox one controller wireless or wired

TheHunterDog, 2017-01-30:
Lakka 
Don't sees a Xbox one controller wired or wireless and in their post they said it's supported so it it's a bug or I need help

Rocklima2, 2017-01-31:
Lakka don't support Xbox One wireless, Lakka suport Xbox One Controller usb cable plugged.

LukStarkiller, 2017-02-16:
Plugged it sees the controller but it doesn't work anyways. Tested with two different xbox one controllers from the last model (the one that comes with the xbox one s), on two different computers with the last lakka build.

Kivutar, 2017-02-17:
OK what hardware are you running Lakka on?

LukStarkiller, 2017-02-17:
PC with booting from an USB with the Live option.

Fisto69, 2017-02-25:
experiencing this as well.    

booted into live usb, doesn't recognize the controller even though i have it plugged in directly via USB

Yaff1e, 2017-05-03:
I'm also seeing this. I've just upgraded my Pi 2 to a Pi 3 and have installed Lakka. I went and bought an Xbox One controller as I didn't have anything else suitable. It's one of the bluetooth S versions (model 1708). Lakka doesn't recognise it.

If I unplug it and replug it back in it vibrates once but that's it, nothing displayed on screen. I also can't start or shutdown Lakka with the controller connected, it just seems to hang.

Kivutar, 2017-05-03:
Please test with the Lakka 2.0 version, I'm using the S version right now and it's working

Yaff1e, 2017-05-12:
Yeap, all working now thanks. Typically I'd installed Lakka and as I didn't have any suitable controllers I ordered a Xbox One controller at the same time. Between placing the order and the controller arriving 2.0 had been released and I'd missed it.

closed 2017-05-12 by Kivutar