libretro/Lakka #636: Add option to enable WIFI AP within WIFI settings

duduke, 2017-01-30:
Hi,

I'm often traveling with my raspberry pi and LAKKA, and while hotel networks are fine most of the time. Having lakka act as an AP for local multiplayer games is awesome.

Using the following command:
**connmanctl tether wifi on <my-ssid> <my-ssid-passphrase>**
Gets an AP up and running.

I know the above relies on the wifi driver supporting AP mode, but at least for the rpi3, it is supported out-of-the-box.

Can you please add this option to the XMB wifi settings?

Kivutar, 2017-02-05:
Very interesting!

duduke, 2017-02-06:
Maybe even enable it by default in new LAKKA installs and together with some mobile phone app (which would mimic a gamepad based off the remote retropad code) we can have an easy "new user" experience which does not require you to attach a keyboard or controller.