libretro/Lakka #637: Compute Stick Drivers

emoryb, 2017-01-31:
The 32 bit generic efi version of Lakka can currently be installed on the Intel Compute Stick gen 1, however, it seems to be lacking support for the built in wifi and bluetooth controllers based on my experience and according to a couple other people on the forums and on Reddit. Also, in my case the sd card reader doesn't seem to work, or at least sd cards won't mount.

gouchi, 2017-01-31:
We need to package [this driver](https://github.com/hadess/rtl8723bs) for the wifi and [this one](https://github.com/lwfinger/rtl8723bs_bt) for bluetooth.

[Source](https://forums.libretro.com/t/intel-compute-stick-wifi/6575)

emoryb, 2017-02-06:
Is there a good tutorial or learning resource where I could learn to do this myself? I'd love to learn how to do it but I don't know where to start.

gouchi, 2017-02-06:
First you can try to [compile Lakka ](http://www.lakka.tv/doc/Compiling-Lakka) then check for example [those drivers](https://github.com/libretro/Lakka/tree/lakka/packages/linux-drivers). The idea is to write [package.mk](http://www.lakka.tv/doc/Packaging-a-libretro-core-for-Lakka/#writing-the-packagemk) to compile the drivers.