libretro/Lakka #638: Low Fps PCSX Rearmed on Pi 3

Gurk232, 2017-01-31:
I am getting extremely low fps from pcsx rearmed on the pi 3, it averages at 24 fps and thus runs extremely slowly. It does this on a variety of tested games.

I believe this has to do with the dynamic_recomplier option, it is not available in the latest available version of the core downloaded via Lakka, manually entering the dynamic_recomplier setting into the config also achieves nothing, I can only assume it was accidently turned off.

But that's only my guess
Running the latest stable version of Lakka on rpi3

And thank you to the Retroarch team for all of the hard work! Can't imagine a world without Retroarch anymore.

closed 2017-02-02 by Gurk232

mayuso, 2017-02-02:
Why did you close this issue?

Did you manage to solve it? How?

Gurk232, 2017-02-03:
Yes, delete the downloaded core from buildbot via samba and lakka will use the included core which works correctly.