libretro/Lakka #639: Why don't use X11 for Generic builds?

mlnlbrt, 2017-02-05:
I know it's better not to use closed source blobs, but I think it's necessary in case of nvidia GPUs. Let's see:

I have GeForce 7600GS and all 2D cores (bettle-psx, snes9x etc.) work fine, but for 3D I'm getting nothing but garbage, flickering and artifacts. I've used Alexandre's fork (see #391) to get Lakka with nvidia 304.125 instead of nouveau and everything works fine.

My point is that 7600GS is a >10 years old card and nouveau still can't handle it properly, at least in terms of 3D. What about newer GPUs?

I think it's a good idea to use X11 for Generic builds, update nvidia packages and cherry-pick some commits from OpenELEC to support both nvidia and nvidia-legacy at the same time. It would allow to support wider range of hardware, as now it seems that Lakka supports intel/amd GPUs and only pretends to support nvidia GPUs. Don't get me wrong, I don't blame anyone. :)

I'm working on this on my own branch, I'll do pull-request when it's ready.

What do you think about the whole thing?

Thanks

gouchi, 2017-02-05:
We are using nouveau driver because we are using DRM/KMS and nvidia driver just [started to implement it](https://devtalk.nvidia.com/default/topic/925605/linux/nvidia-364-12-release-vulkan-glvnd-drm-kms-and-eglstreams/). For the time being, can you try to make a test with [Lakka (LibreELEC version)](http://le.builds.lakka.tv/Generic.x86_64) if you reproduce your issue.

Thank you.

mlnlbrt, 2017-02-07:
I didn't know such builds exist, thanks. :) I'll give it a try soon.

I'm just curious, why are we using DRM/KMS instead of X11? I'm aware that it's a cleaner approach and stuff, but are there any other reasons?

I'm still thinking that nvidia blobs will give better experience than nouveau, especially for cores like GlupeN64 or reicast.

Kivutar, 2017-02-07:
DRM/KMS means less bloat and results in less input lag. The cleaner approach, as you said.

However, just enabling X11 in the option file lets you compile your own image with X11. So the community could take care of producing X11 builds.

closed 2017-04-03 by Kivutar