libretro/Lakka #641: Stuck on Flower Logo - NVIDIA Corporation NV17 [GeForce4 MX 440]

cirelli94, 2017-02-13:
- Which version of Lakka are you using?
Lakka (unofficial) Version: devel-20161221183352-r21729-g274b40f
Lakka git: 274b40f7a5f7a880309a4fe8765e69ac8d8153e1

- What system hardware are you using?
I think the problem is due to:
Lakka:~ # lspci -nn | grep VGA
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation NV17 [GeForce4 MX 440] [10de:0171](rev a3)

- What happen?
Both installed and live version stuck on flower logo

retroarch -v:

RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2
Built: Dec 21 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 274b40f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 16 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1366x768) 1366 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (720x400) 720 x 400, 70 Hz
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/nouveau_vieux_dri.so: cannot open shared object file: No such file or directory
failed to load driver: nouveau_vieux
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No suchfile or directory
failed to load driver: kms_swrast
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
failed to load swrast driver
RetroArch [WARN] :: [KMS]: Couldn't create GBM device.
RetroArch [WARN] :: [DRM]: Couldn't get device resources.
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 16 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1366x768) 1366 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (720x400) 720 x 400, 70 Hz
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No suchfile or directory
failed to load driver: kms_swrast
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
failed to load swrast driver
RetroArch [WARN] :: [KMS]: Couldn't create GBM device.
RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).
Segmentation fault

I'm now updating with lakka-update.

gouchi, 2017-02-14:
Your graphic is old so it seems we need to enable nouveau_vieux_dri in mesa.

cirelli94, 2017-02-15:
So should I change
`DRI_DRIVERS="$DRI_DRIVERS"`
in 
`DRI_DRIVERS="$DRI_DRIVERS,nouveau_vieux_dri"` ?
And compile.

gouchi, 2017-02-15:
Yes should be like that.

HIH.

cirelli94, 2017-02-15:
Well, I did it but the problem is the same!
1. I had a problem compiling libcap. I temporary fixed it with adding #include <stdio.h> in _caps_output.gperf and modifiy  __cap_lookup_name (register const char *str, register size_t len) to  __cap_lookup_name (register const char *str, unsigned int len)

2. All is the same, i ssh-ed and the output of retroarch -v is unchanged!

Thanks for your help mate

gouchi, 2017-02-15:
To be sure can you try to recompile mesa package 
```
DISTRO=Lakka PROJECT=Generic ARCH=i386 scripts/clean mesa
DISTRO=Lakka PROJECT=Generic ARCH=i386 scripts/build mesa
```
For 64bit please change ARCH to x86_64

Thank you for your tests.

Mwea, 2017-02-17:
Try to edit your retroarch.cfg file 
Disable the video_threaded option and reboot. 
It may fix the problem.

cirelli94, 2017-02-17:
@gouchi Ok, I did a mistake and I didn't recompiled mesa package. Now I cleaned first. But:
```
DRI_DRIVERS="$DRI_DRIVERS,nouveau_vieux"
configure: error: classic DRI driver 'nouveau_vieux' does not exist

```
```
DRI_DRIVERS="$DRI_DRIVERS,nouveau_vieux_dri"
configure: error: classic DRI driver 'nouveau_vieux_dri' does not exist

```
```
DRI_DRIVERS="$DRI_DRIVERS,nouveau_vieux_dri.so"
configure: error: classic DRI driver 'nouveau_vieux_dri.so' does not exist

```

Meanwhile
```
DRI_DRIVERS="$DRI_DRIVERS,nouveau"
```
is compiling right now.

Nouveau docs says:
The Gallium3D-based drivers are all compiled into nouveau_dri.so. The correct library (nouveau_vieux_dri.so vs. nouveau_dri.so) and the right driver are automatically selected runtime, if they are available.
So I hope I'm doing right now.
Tried: it's still the same.

@Mwea So I'm sshed now and I change storage/.config/retroarch/retroarch.cfg
Thank @Mwea but this doesn't help!

I think I still haven't installed the right drivers configuration at compile time.

gouchi, 2017-02-18:
Yes I made a mistake it should be `DRI_DRIVERS="$DRI_DRIVERS,nouveau"`

Can you check in the configure that you got 
`checking for NVVIEUX... yes`