libretro/Lakka #642: Orange Pi Plus stuck on flower boot screen

hooby3dfx, 2017-02-15:
- Which version of Lakka are you using?
Latest nightly:
http://mirror.lakka.tv/nightly/H3.opiplus.arm/Lakka-H3.opiplus.arm-devel-20161221193357-r21729-g274b40f.img.gz
- What system hardware are you using?
Orange Pi Plus
- What did you do?
Boot the system after loading above image on sandisk sd card with win32diskimager. This sd card successfully boots other os images.
- What did you expect to happen?
Boot and then launch retroarch
- What happened instead?
Stuck on flower screen with text "sh: can't access tty: job control turned off"

Kivutar, 2017-02-17:
Ok, this error is usually a flashing error. Can you try to reflash the same image or try the LibreELEC based one here please? http://le.builds.lakka.tv/

hooby3dfx, 2017-02-19:
Thanks - I was successful with this one: 
http://le.builds.lakka.tv/H3.opiplus.arm/Lakka-H3.opiplus.arm-8.0-devel-20170218165610-r25669-gef3bc72.img.gz

closed 2017-02-19 by gouchi