libretro/Lakka #643: PPSSPP - Gran Turismo not working

Skirmitt83, 2017-02-15:
I'm not able to launch Gran Turismo with the PPSSPP core.
I found an old thread pointing to a font issue here: https://github.com/libretro/Lakka/issues/443

This issue isn't solved in the latest version. What could be wrong ?
De Windows version of PPSSPP is able to run the game perfectly.

Kivutar, 2017-02-15:
Hello, did you check our documentation about BIOS? You have to download a font and put it in a specific directory.

Skirmitt83, 2017-02-15:
Hi,
Yes the bios is ok because I can play all the other games.
I have the font here but where do I put it ? PPSSPP says the system folder but is this the same system folder from Lakka ? I guess not because I already tried that out.

Kivutar, 2017-02-15:
This font is needed for the saving interface. Some games require it, some will just crash when you try to save, and some won't save anything and won't need the font.
All the informations are on the BIOS doc page.

Skirmitt83, 2017-02-17:
Found it, will try this next week, will keep you updated

Skirmitt83, 2017-02-20:
I added the flash0 and lang folder with all its content. Now I can get into the menu but when I try to load a race the game crashes.