libretro/Lakka #644: stuck at flower splash screen

sketchy1, 2017-02-16:
so im trying to install lakka onto this netbook i have, but i cant seem to get passed the flower splash screen. upon closer inspection, i noticed it stops at 
"reached retroarch" (wording might be wrong)

this happens when im trying to boot from live cd. i dont exactly want to install it until i can confirm it works. if it helps, im using this pc:

https://www.cnet.com/products/asus-eee-pc-disney-8-9-atom-n270-1-gb-ram-160-gb-hdd-mk90hblu002x/specs/

also, im using the latest nightly, here:
http://sources.lakka.tv/nightly/Generic.i386/
named:
Lakka-Generic.i386-devel-20161221183902-r21729-g274b40f-efi.img.gz

Kivutar, 2017-02-17:
Certainly a graphic card that we don't support yet.

@gouchi can you confirm please?

gouchi, 2017-02-18:
Hi,

Is it possible to get lakka system and graphic [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

It seems your card is using intel gma 950 so it should be ok.

Your netbook supports UEFI boot ?

sketchy1, 2017-02-19:
yes, my system dows in fact have uefi boot support, but how can i dump the logs?

gouchi, 2017-02-19:
"The file log.txt is now stored in the home directory. You can now copy the log file off of the Lakka system via an SCP file transfer."