libretro/Lakka #645: GB-BKi3A-7100 Support

link99x, 2017-02-18:
Hi,
the new Gigybayte Brix GB-BKi3A-7100 hangs on the boot screen. (Intel Kaby Lake Gen - HD Graphics 620).
Error Log:

RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AVX2 AES 
Built: Dec 21 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 274b40f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
**RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.**
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).
Segmentation fault

>> RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.
So it seems that a driver for the Kaby Lake Generation is missing. It would be great, when newer releases will support the new intel generation.

Thanks for your help
lg

gouchi, 2017-02-19:
Can you try if you reproduce the issue with [Lakka (LibreELEC version)](http://le.builds.lakka.tv/Generic.x86_64/) ? If you do, please open an [issue](https://github.com/lakkatv/Lakka-LibreELEC/issues).

Thank  you for your test.

link99x, 2017-02-19:
The LibreELEC version works well, no problems detected. Also the wifi and the bluetooth module is working. Thanks a lot for the tip.
br

closed 2017-02-19 by gouchi