libretro/Lakka #648: Switch Lakka to be forked from LibreELEC instead of OpenELEC?

Hedda, 2017-02-21:
Would it not be a good idea for Lakka to switch fork to LibreELEC instead of forking OpenELEC?

Reason for this being that OpenELEC now days only have one single active developer, while LibreELEC (a fork of OpenELEC) have over 30 active developers.

In case you did not hear, OpenELEC though not fully dead have been totally abandonded by all its former developers with the exception of its original founder.

All developers from OpenELEC have instead forked the code and started a new project named LibreELEC / LibreELEC.tv

https://github.com/LibreELEC/LibreELEC.tv

https://libreelec.tv

Kodi team / developers and more former OpenELEC supporters now recommend that all users switch over to LibreELEC.

hizzlekizzle, 2017-02-21:
Yep, it's in-process.

Kivutar, 2017-02-21:
The repo: https://github.com/lakkatv/Lakka-LibreELEC
The builds: http://le.builds.lakka.tv/

closed 2017-02-21 by Kivutar