libretro/Lakka #652: Issue Scanning ROMs in a folder with .7z (7Zip) files

josefprado, 2017-03-03:
I found out that Lakka is able to scan a ROM inside a .7z file, but its not able to do this when we select a folder with lots of ROMs compressed with 7Z, it seems just to ignore then.

This appear to be not a big issue, since we can just convert our .7z files to .zip and Lakka would be able to read it, but this is not the point.

7Zip is far more effective in compressing the ROMs than any .ZIP tool, and having our ROMs 7Zipped instead of only Zipeed is a huge gain, specially if you are running Lakka on a Raspberry or similar solution, where you commonly save your ROMs inside a SD Card, with a limitted space.

Just for comparission, my entire Genesis/Megadrive ROM collection have 639Mb when each Game ROM Set is 7Zipped, but i grows to almos 4Gb when I do exactly the same, but using the common Zip format, even with Maximum compression. Remember that we are talking about a smal Megadrive ROM set, can you Imagine the size difference when we talk about a Final Burn or a MAME collection?

So I hope you guys could solve the issue that is amkking Lakka ignore the 7Zip files when scanning a folder,  since this would allow us to have a bigger collection using a single SD card.

bparker06, 2017-03-03:
I believe you may be hitting the RetroArch bug we had where you need MAME or FBA downloaded in order to scan archives correctly. This was recently fixed in the git master but Lakka might have an older build. Can you try making sure you have one of those cores, as well as your core info files and databases all up to date?

josefprado, 2017-03-03:
Hi Parker,

I do have FBA running, since I´m playing the FBA games (and it can scan folders normally, since the ROMs are ZIPPED, not 7ZIPPED).

Is this the RetroArch bug you are mentioning?

Anyway, tonight I will make sure I have the latest FBA core installed, and try it again, but I dont think this is the issue, since the problem happens in all kind of ROM, MAME, FBA, MegaDrive, NES, SNES... It just ignores the .7z files when you ask to Scan a folder (besides being able to correctly scan .7z the file and read the ROM if you ask to "Scan File")

Regards,

Josef

chongtin, 2017-03-12:
I have the same issue. The work around is to convert all 7z files to zip files, copy to the rom direct, and than have it scan. Afterwork copy those 7z files to that same directory, and finally edit the playlist file under Playlists directory. Open it with a text editor, and replace all .zip to .7z

chongtin, 2017-03-16:
This problem seems fixed in the new le version of nightly build

josefprado, 2017-03-18:
This will be great, since 7Zip makes the file much much smaller, allowing us to have more files in th SD card and also making is fast to copy over the network.

Do you know how much time does it take for the nightly changes to become available in a stable build?

chongtin, 2017-03-18:
Sorry, I am just a user, not a developer. If you are interested in trying it, you can download it from here.
http://le.builds.lakka.tv/

If I were you, I will try it with another media/sd card than the one you are currently using. Since nightly build are supposed to be not stable.

josefprado, 2017-05-02:
Juste tested in new Version 2.0 stable and it worked fine.

closed 2017-05-02 by josefprado

mikel632, 2017-05-11:
I have version 2.0 that I just installed on rp3 and when I go to scan a directory it doesn't show any items. I hit scan anyway and then it just freezes. My roms (snes) are .zip and from no-intro rom set. I can add each one manually but I'd rather not do that. I went to load core and chose FBA and also Mame2003 but that did not help at all. Any ideas?

Kivutar, 2017-05-12:
Scan folder per folder, it may freeze if you try to scan too much stuff like arcade+CDs.