libretro/Lakka #654: X-One Day One Edition Controller Issue

josefprado, 2017-03-05:
Hi!

I´m trying to use my X-Box one Day One Edition controler, but for some reason i just does not work with Lakka.

Lakka recognizes it, says its configured, but i just dont work. If I plug a common X-Box One controller (one I bougth separately from the console) it works perfectly.

I think its a bug, but if I´m missing any step or doing something wrong, please let me know.

Kivutar, 2017-03-05:
Hello, can you tell us the vid:pid of the controller please?

You can know it with the lsusb command in ssh.

josefprado, 2017-03-05:
Hi,

Theu seem to have the same one:

```
Lakka:~ # lsusb
Bus 001 Device 005: ID 045e:02d1 Microsoft Corp. XBOX One Controller for Windows
Bus 001 Device 004: ID 045e:02d1 Microsoft Corp. XBOX One Controller for Windows
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ #
```
Day One Edition in on the botton Left USB port.

What I did notice is that the standard controller get the big X "on", but the Day One Edition, it keeps Off (If i touch it it turn on my Xbox One :)

josefprado, 2017-03-11:
Hi, 

I just got a "LSUSB -V" on the same controller, maybe this can help you guyes better:

`Bus 001 Device 004: ID 045e:02d1 Microsoft Corp. XBOX One Controller for Windows
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71
  bDeviceProtocol       208
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x02d1 XBOX One Controller for Windows
  bcdDevice            2.03
  iManufacturer           1 Microsoft
  iProduct                2 Controller
  iSerial                 3 7EED82B91982
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           96
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71
      bInterfaceProtocol    208
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71
      bInterfaceProtocol    208
      iInterface              0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71
      bInterfaceProtocol    208
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71
      bInterfaceProtocol    208
      iInterface              0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71
      bInterfaceProtocol    208
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
`

Kivutar, 2017-04-03:
Ok, I think you will need to switch to our new beta builds here: http://le.builds.lakka.tv/
We support more Xbox One controllers on these new images.

josefprado, 2017-05-02:
I just tested and It´s working fine on Lakka 2.0 Stable.

closed 2017-05-02 by josefprado