libretro/Lakka #656: Lakka locked at flower Boot screen

Alechu26, 2017-03-08:
So I discovered this distro just yesterday while testing some distros on an old PC I revived a day ago. Things seemed to be fine until I tried to Boot Lakka, it just locks at the flower screen. Tried both 32 bit and 64bit version with no success. The 32bit version Will give some flickering on the screen while trying to bot on the flower screen. This didn't happen on the 64bit version tho. 
The System i'm running now has:
AMD Sempron 2600+ 1.6Ghz single core
1.5GB DDR400 RAM
ATI Radeon 9250 with 256MB Vram
I know it's pretty old but I thought that Lakka should work. Maybe it doesnt like the Graphics card, it's kinda shitty. 
So, anyway to try and fix this? Thanks!

leolite1, 2017-03-18:
I ran into a similar issue where the video was a different display a while back ( #434). I used the [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/)  for debugging.

evbo, 2017-05-11:
I had this same issue:

1. I installed Lakka 64-bit from here: http://www.lakka.tv/get/windows/generic/

    1. Lakka displayed fine on my laptop monitor, but it wouldn't display on my external HDMI connected monitor

    1. So I followed these steps to set the default monitor index to "2" instead of "0": https://github.com/libretro/Lakka/issues/465

    1. But, that didn't work on its own, so then I tried this and it worked: https://github.com/libretro/Lakka/issues/288

1. But, then I needed support for xbox one s controllers. So I started over, installing 2.0 instead: http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.0.tar

    1. Same display issues, so I followed the same steps as above

But, this time, nothing seems to work. Since Lakka boots expecting monitor index 2, it just hangs and the external monitor keeps flickering but no solid connection is sustained. My laptop monitor just keeps showing the flower.

relevant info:

```
Lakka:~ # for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
HDMI-A-1: connected
LVDS-1: connected
VGA-1: disconnected
```

```
Lakka:~ # lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Complex [1022:1705]
00:01.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Sumo [Radeon HD 6480G] [1002:9648]
00:01.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] BeaverCreek HDMI Audio [Radeon HD 6500D and 6400G-6600G series] [1002:1714]
00:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:1709]
00:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:170a]
00:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:170b]
00:11.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7804]
00:12.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:12.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:13.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:13.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:780b] (rev 13)
00:14.2 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] FCH Azalia Controller [1022:780d] (rev 01)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:780e] (rev 11)
00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] FCH PCI Bridge [1022:780f] (rev 40)
00:16.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:16.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 0 [1022:1700] (rev 43)
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 1 [1022:1701]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 2 [1022:1702]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 3 [1022:1703]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 4 [1022:1704]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 6 [1022:1718]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 5 [1022:1716]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 7 [1022:1719]
01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter [10ec:8176] (rev 01)
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8101/2/6E PCI Express Fast/Gigabit Ethernet controller [10ec:8136] (rev 05)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5209 PCI Express Card Reader [10ec:5209] (rev 01)
03:00.1 SD Host controller [0805]: Realtek Semiconductor Co., Ltd. RTS5209 PCI Express Card Reader [10ec:5209] (rev 01)
```