libretro/Lakka #660: Steam integration

jeapostrophe, 2017-03-14:
I would like to be able to use Steam on my Lakka box. I have a few ideas on how to make it work (ranging from easier to harder):

1. Support dual-booting to something else. This would allow the user to set up either Windows or Steam OS. Ideally, this would come with controller support in the bootloader or boot loader scripts to control booting. (For example: Always boot into Lakka, but Lakka can drop a flag in the boot partition to go into the other OS, which is a signal to only do it once and then go back to Lakka. Thus Lakka can always be in charge.)

2. Support running Steam for Linux on x86 builds. I suspect that this will be painful, but do-able. Lakka would launch Steam and when it quit, it would give control back to Lakka.

3. Use Moonlight or similar API to directly steam into Lakka.

jeapostrophe, 2017-03-14:
@Kivutar I know you've commented before that you don't plan on supporting this, so I don't expect you will spend your time on this. But I would prefer there to be one issue to talk about it on the repo. I intend to try to implement approach 1 and then try other approaches slowly.

jeapostrophe, 2017-03-14:
1.  Most boot-loaders support this easily:

GRUB: This can be done fairly easily with the `grub-reboot` option documented here: https://wiki.debian.org/GrubReboot

systemd-boot: "For systemd-boot, use systemctl reboot _name_, where _name_ is the name of a boot entry that will be selected at next boot and only the next boot. "

syslinux: "for the Syslinux family the --once "cmd" option."

jeapostrophe, 2017-03-14:
3. There is a Moonlight addon for LibreELEC already:

https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/addon-depends/moonlight-embedded/package.mk

And here is a guide for setup:

http://www.makeuseof.com/tag/stream-steam-games-raspberry-pi-diy-link-box/

So, the path here would be to provide a convenient menu interface in Retroarch for working with Moonlight

Kivutar, 2017-03-15:
I would prefer a moonlight libretro core :)

Kivutar, 2017-04-03:
There is now a bounty on this https://github.com/libretro/libretro-meta/issues/76