libretro/Lakka #664: "ALSA: Failed to initialize..." on RPI3 with Lakka RC3

ZeikJT, 2017-03-24:
I had been using the last stable version of Lakka (`20161221184608-r21729-g274b40f`, 21-Dec-2016) on my Raspberry Pi 3 but decided to try out the RC3 build to see what I could expect when it goes stable soon.

I downloaded `Lakka-RPi2.arm-2.0-rc3.tar` but when it's installed the hdmi audio stops working. I've tried combinations of `/flash/config.txt` changes including `hdmi_drive=2` and `hdmi_force_edid_audio=1` but no luck.

When I run `aplay -L` on the December stable version I see:
```
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
```

But, when I run the same command on the RC3 version all I see is:
```
null
    Discard all samples (playback) or generate zero samples (capture)
```

I logged the retroarch startup for both builds. The stable build is clean of audio problems, but on the RC3 build I see some errors:
```
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/conf.c:4371:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/conf.c:4371:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/conf.c:4371:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/conf.c:4850:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0-rc3/alsa-lib-1.1.2/src/pcm/pcm.c:2450:(snd_pcm_open_noupdate) Unknown PCM default
RetroArch [ERROR] :: ALSA: Failed to initialize...
RetroArch [ERROR] :: Failed to initialize audio driver. Will continue without audio.
...
RetroArch [ERROR] :: failed_to_start_audio_driver
RetroArch [ERROR] :: failed_to_start_audio_driver
```

ZeikJT, 2017-03-25:
After doing more testing, looks like adding `dtparam=audio=on` to my config.txt makes everything work again. Was this part of a config in the previous 1.x releases and removed in 2.0?

In any case, closing my issue.

closed 2017-03-25 by ZeikJT

Kivutar, 2017-03-26:
Isn't it there? https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/packages/tools/bcm2835-bootloader/files/3rdparty/bootloader/distroconfig.txt#L22

ZeikJT, 2017-03-26:
I think I understand what might have happened. I was updating by putting the new tar file in the updates folder and then rebooting. The config from the stable December release did not have the line `include distroconfig.txt` and it seems like this update process avoids overwriting the `config.txt` file?

Is that expected?

Kivutar, 2017-03-26:
Yes, we ask the users of the december build to do an installation from scratch when they switch to lakka 2.0.

ZeikJT, 2017-03-27:
Gotcha, thanks for the info!

I updated after seeing the [news post](http://www.lakka.tv/articles/2017/03/13/lakka-20-rc3-ships-retroarch-150-and-brings-rpi-zero-w-support/) and it didn't mention that tidbit. Hopefully this issue helps people who might make the same mistake I did.