libretro/Lakka #665: Lakka 2.0 RC3 not starting on Intel NUC DN2820FYKH

aboutafter, 2017-03-25:
Lakka 2.0 RC3 doesn't start on the Intel NUC DN2820FYKH when booting from the USB drive. 

Selecting the USB drive where Lakka is to boot, it simply jumps straight to the PC's grub menu because it doesn't seem to recognize any booting partitions on it. 

This same USB drive works correctly on my desktop PC (Intel i5-4670K and Nvidia GTX 770) and on my Thinkpad x230 laptop (Intel i7-3520M).

gouchi, 2017-03-26:
Please open an issue to the [correct issue tracker.](https://github.com/lakkatv/Lakka-LibreELEC/issues)

Thank you.

Kind regards,

aboutafter, 2017-03-26:
Uups, my bad. This is the new link: https://github.com/lakkatv/Lakka-LibreELEC/issues/86

closed 2017-03-26 by Kivutar