libretro/Lakka #668: Add Support for Mike Chi's RPi 240p Composite/S-Video/Component Hat

kevinfishburne, 2017-04-03:
Using the latest Lakka image on my Raspberry Pi 3 Model B it fails to output video using Mike Chi's RPi hat. Raspbian has no such issues, so it has something to do with firmware or other files in the /boot partition. Mike's config file is attached: [config.txt](https://github.com/libretro/Lakka/files/889147/config.txt)

I have v2 of his hat prototype, which may be ordered here: http://www.retrotink.com/

Links to and discussion of the hat can be found here: https://discord.gg/zqUdBn6

Direct github link here: https://github.com/libretro/libretro-meta/issues/73

kevinfishburne, 2017-04-04:
Just tried it with http://le.builds.lakka.tv/RPi2.arm/Lakka-RPi2.arm-2.0-rc4.tar and I'll be damned if it isn't working perfectly now. Closing as fixed. Thank you Team Lakka. :)

closed 2017-04-04 by kevinfishburne

zackforbing, 2017-06-24:
@kevinfishburne were you using Mike's custom Lakka install? I am using that too and it won't load Lakka, just boots into retroarch. was this how yours was functioning?

hizzlekizzle, 2017-06-24:
@zackforbing 
Lakka always boots directly into RetroArch. Is there something else you were expecting?

kevinfishburne, 2017-06-24:
There was some kind of bug that was fixed with the 2.0-rc4 build. I was using the vanilla Lakka image, not a custom build. And Lakka does boot into Retroarch. It's Retroarch on top of LibreELEC with some customizations to make it more turnkey.