libretro/Lakka #670: Vakoss joypad not working

realmario84, 2017-04-07:
Hello
I bought cheap wireless joypad (Vakoss GP-3925BK) and it's not working with Lakka. I' ve checked it on Ubuntu, Windows 10 and RetroPie and it works. In version Lakka-RPi2.arm-devel-20161221184608-r21729-g274b40f it was seen as ShanWan PC/PS3/Android but I couldn't set any binding. No button was recognized. Not I run Lakka 2.0-rc4 and it's even worst because it not recognize it in Lakka at all. Here are some debug info:

Lakka:/ # lsusb 
**Bus 001 Device 007: ID 20bc:5500**  
Bus 001 Device 008: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# dmesg
408.098203] usb 1-1.4: new full-speed USB device number 9 using dwc_otg
[  408.192303] usb 1-1.4: New USB device found, idVendor=2563, idProduct=0575
[  408.192320] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  408.192327] usb 1-1.4: Product: PC/PS3/Android
[  408.192334] usb 1-1.4: Manufacturer: ShanWan
[  408.204955] input: ShanWan PC/PS3/Android as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:2563:0575.0008/input/input3
[  408.205702] hid-generic 0003:2563:0575.0008: input,hidraw1: USB HID v1.10 Gamepad [ShanWan PC/PS3/Android] on usb-3f980000.usb-1.4/input0
[  408.255679] usb 1-1.4: USB disconnect, device number 9
[  408.528172] usb 1-1.4: new full-speed USB device number 10 using dwc_otg
[  408.623809] usb 1-1.4: New USB device found, idVendor=20bc, idProduct=5500
[  408.623826] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  408.623833] usb 1-1.4: Product: PC/PS3/Android
[  408.623840] usb 1-1.4: Manufacturer: ShanWan

I've checked that on Ubuntu it is using driver usbhid and betop (module hid_betopff) and on Lakka I don't see such modules.

Lakka:/ # lsmod 
Module                  Size  Used by
hci_uart               15605  1
btbcm                   5315  1 hci_uart
bluetooth             291781  21 hci_uart,btbcm
8021q                  16707  0
brcmfmac              181893  0
brcmutil                4893  1 brcmfmac
cfg80211              246778  1 brcmfmac
rfkill                 13757  5 bluetooth,cfg80211
bcm2835_gpiomem         2652  0
fixed                   2407  0
overlay                31427  5

Any help will be appreciated.

Sobient, 2017-04-10:
There are many chinese clones that don't work well, mostly because their bluetooth module doesn't identify them properly (there are many threads around the web on this), so Lakka certainly isn't the first. I've had a clone Dualshock 3 that never could connect via bluetooth. I would suggest [reading about inputs on the Wiki](http://www.lakka.tv/doc/Input-settings/), especially the section of manually binding your inputs. Remember there are tons of clones out there and supporting them all is a behemoth-task. 

There are options though; You can consider supporting Lakka (Patreon, sending them hardware, etc.) to help speed up support for "mis-behaving hardware".

realmario84, 2017-04-10:
This joypad is connected via wireless USB nano adapter, not bluetooth. I close thread because I installed RetroPie and it have hid_betopff module so joypad works.

closed 2017-04-10 by realmario84

Demkan, 2017-06-22:
Hello,
I have the same gamepad with 2.4G wireless USB adapter with the same problem as described by realmario84. Lakka shows me two messages when connecting the nano adapter: 1 that the gamepad ist not configured, and 2 that it is disconnected from port 0. I have found the source code of the module hid_betopff at http://elixir.free-electrons.com/linux/latest/source/drivers/hid/hid-betopff.c#L41

Is it possible to add this module to Lakka?

Thanks for any help

Kivutar, 2017-06-22:
Hello, the new tracker is there: https://github.com/libretro/Lakka-LibreELEC/issues