libretro/Lakka #672: WII remote not connecting again with Lakka

samdawat, 2017-04-14:
so i all ready successfully connected my Wii remote the first time with Lakka however after i rebooted it it would no longer connect with my Wii remote, i have pressed literally all the buttons hoping for it to connect but it wont here is what it says when using SSH:

Lakka (unofficial) Version: devel-20161221184608-r21729-g274b40f
Lakka git: 274b40f7a5f7a880309a4fe8765e69ac8d8153e1
Lakka:~ # systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor pr                      eset: enabled)
   Active: active (running) since Fri 2017-04-14 17:15:12 UTC; 4min 38s ago
 Main PID: 566 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─566 /usr/lib/bluetooth/bluetoothd

Apr 14 17:15:12 Lakka systemd[1]: Started Bluetooth service.
Apr 14 17:15:12 Lakka bluetoothd[566]: Bluetooth daemon 5.37
Apr 14 17:15:12 Lakka bluetoothd[566]: Starting SDP server
Apr 14 17:15:12 Lakka bluetoothd[566]: kernel lacks bnep-protocol support
Apr 14 17:15:12 Lakka bluetoothd[566]: System does not support network plugin
Apr 14 17:15:12 Lakka bluetoothd[566]: Failed to open RFKILL control device
Apr 14 17:15:12 Lakka bluetoothd[566]: Bluetooth management interface 1.9 i...ed
Apr 14 17:15:12 Lakka bluetoothd[566]: Failed to obtain handles for "Servic...ic
Hint: Some lines were ellipsized, use -l to show in full.
Lakka:~ # bluetoothctl
[NEW] Controller B8:27:EB:1A:05:B5 BlueZ 5.37 [default]
[NEW] Device 00:5A:19:00:3F:2F Nintendo RVL-CNT-01
[NEW] Device A4:C0:E1:2E:85:55 Nintendo RVL-CNT-01
[bluetooth]# connect A4:C0:E1:2E:85:55
Attempting to connect to A4:C0:E1:2E:85:55
Failed to connect: org.bluez.Error.NotReady
[bluetooth]# connect A4:C0:E1:2E:85:55
Attempting to connect to A4:C0:E1:2E:85:55
Failed to connect: org.bluez.Error.NotReady

(Note: im using the current build of Lakka (last version in april 2017 dont know whats it called))

samdawat, 2017-04-14:
BTW in using the raspberry PI3

ervee, 2017-05-14:
Check the of commands within bluetoothctl in issue #578 and see if that fixes it for you.

samdawat, 2017-05-15:
it worked thx