libretro/Lakka #677: Dell Inspiron 1150 bootloop

Hurthtet, 2017-05-09:
Hello, I am trying to install Lakka on an old Dell Inspiron 1150. The graphics processor is an "Intel Extreme Graphics 2" chip, video controller "Intel 852GM/GME/GMV". 
   First, I attempted to use the Live USB image, "Lakka-Generic.i386-2.0.img". The install appeared to work, but upon trying to reboot, I was given the error 
                                "/usr/bin/installer: line 479: reboot: Input/output error".
   After this, when the PC was turned on, it would boot up and display the Lakka logo, and then get stuck in an infinite loop of displaying the logo, going to blackscreen, and redisplaying the logo. 
   Next, I tried installing using the December 21 2016 nightly build, which was the most recent one I found, "Lakka-Generic.i386-devel-20161221183902-r21729-g274b40f.tar". This time, I was not brought to any install screen, but the PC immediately resumed it's loop of logo, blackscreen, logo.
If someone could help me out with this, it would be much appreciated.

Kivutar, 2017-05-09:
It looks like RetroArch is not able to start. I'm not sure if we can support "Intel 852GM/GME/GMV".

@gouchi ?

gouchi, 2017-05-09:
I think it should be supported but maybe they removed support for old intel cards ?

Is it possible to get [graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/) ?

You can try to boot with (hit tab on the boot screen)
`live textmode`

Hurthtet, 2017-05-09:
When I boot with `live textmode`, and enter the command `lspci -nnk | grep -A 3 VGA `, I get:
00:02.0 VGA compatible controller [0300]: Intel Corporation 82852/855GM Integrated Graphics Device [8086:3582] (rev 02) Subsystem: Dell 82852/855GM Integrated Graphics Device [1028:017f]
Kernel driver in use: i915
00:02.1 Display Controller [0380]: Intel Corporation 82852/855GM Integrated Graphics Device [8086:3582] (rev02)

gouchi, 2017-05-11:
Also is possible to get [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/) ? I forgot to ask before.

You can try to boot with (hit tab on the boot screen) to connect by [ssh
](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#ssh)`live ssh`

Thank you.

Hurthtet, 2017-05-12:
Here it is, 

Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [WARN] :: screenshot_directory is not an existing directory, ignoring ...
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1024x768.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x9912050.
RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 852GM/855GM x86/MMX/SSE2.
RetroArch [INFO] :: [GL]: Version: 1.3 Mesa 13.0.6.
RetroArch [INFO] :: GL: Using resolution 1024x768
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: Using font rendering backend: bitmap.
RetroArch [ERROR] :: GL: Invalid enum.
RetroArch [ERROR] :: Cannot open video driver ... Exiting ...
RetroArch [ERROR] :: Fatal error received in: "init_video()"
RetroArch [ERROR] :: This core requires a content file.

gouchi, 2017-05-13:
Can you try to edit [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/) and change
`menu_driver = "xmb"` to `menu_driver = "rgui"`
Nano shortcuts (CTRL+o) to save and (CTRL+x to quit).

Hurthtet, 2017-05-14:
I've made the change, but the boot screen still flashes on and off endlessly. If it helps, after I saved the config file and ran `retroarch`, I was given the error `Segmentation fault (core dumped)`.

gouchi, 2017-05-14:
I suspect your graphic card is too old to be used with Retroarch/Lakka.

If you can try a live cd of another distribution  and type 
`glxinfo -B`

It should give us information about opengl version.

Hurthtet, 2017-05-16:
Using an Ubuntu live usb, I tried `glxinfo -B` and got "Unknown option '-B'". Trying `glxinfo -b` just printed "32".

gouchi, 2017-05-16:
```
glxinfo -h
Usage: glxinfo [-v] [-t] [-h] [-b] [-l] [-s] [-i] [-display <dname>]
	-display <dname>: Print GLX visuals on specified server.
	-i: Force an indirect rendering context.
	-B: brief output, print only the basics.
	-v: Print visuals info in verbose form.
	-t: Print verbose table.
	-h: This information.
	-b: Find the 'best' visual and print its number.
	-l: Print interesting OpenGL limits.
	-s: Print a single extension per line.
```

Hurthtet, 2017-05-16:
That's odd, I'm missing the -B option entirely. I'll try a newer version.

Hurthtet, 2017-05-19:
I'm having a hard time finding anything with `glxinfo -B`, what should I use?

gouchi, 2017-05-19:
glxinfo  | grep OpenGL ?